ControlSystemStudio/cs-studio #1885: cs-studio/applications: mvn verify Error

myhippo, 2016-08-30:


willrogers, 2016-08-30:
The cs-studio build is currently broken for the reasons described here: https://github.com/ControlSystemStudio/cs-studio/issues/1859.

I hope that this will be resolved by the end of the week.

willrogers, 2016-08-30:
Hold on, this should already have been fixed.  You are at 30e366b, but `master` is now at 9878102.

Try `git pull origin master` and then build again.

kasemir, 2016-08-30:
Yes, building from the 'master' branch should mostly work.
Of course there are several ways to do this.
In short, my recipe for the SNS products is here: https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build

There is one minor issue, see https://github.com/ControlSystemStudio/cs-studio/pull/1884: We recently deprecated org.csstudio.platform.libs.epics.ui. org.csstudio.vtype.pv.ui offers that functionality for those who want to configure EPICS CA from the preference UI, but to actually get that new plugin into the target platform, you need that pull request.

kasemir, 2016-08-30:
The pull request mentioned in my last comment has been merged.
The complete chain of
 diirt,
 maven-osgi-bundles,
 cs-studio-thirdparty,
 cs-studio/core,
 cs-studio/applications,
 org.csstudio.sns
builds OK in that order from master.
Remaining issues may be in some product files which still need to be updated, but the original error mentioned for this issue report should be resolved.

claudio-rosati, 2016-08-31:
Not yet working for me: cs-studio is still broken.

[INFO] Resolving class path of MavenProject: org.csstudio:org.csstudio.platform.libs.hibernate.ui:1.0.0-SNAPSHOT @ <https://jenkins02.esss.lu.se/job/CS-Studio%20Original%2000500%20-%20cs-studio/ws/cs-studio/core/platform/platform-plugins/org.csstudio.platform.libs.hibernate.ui/pom.xml>
[ERROR] Internal error: java.lang.IllegalArgumentException: bundleLocation not found: <https://jenkins02.esss.lu.se/job/CS-Studio%20Original%2000500%20-%20cs-studio/ws/.m2/p2/osgi/bundle/org.csstudio.platform.libs.hibernate/1.0.0.201510091318/org.csstudio.platform.libs.hibernate-1.0.0.201510091318.jar> -> [Help 1]
org.apache.maven.InternalErrorException: Internal error: java.lang.IllegalArgumentException: bundleLocation not found: <https://jenkins02.esss.lu.se/job/CS-Studio%20Original%2000500%20-%20cs-studio/ws/.m2/p2/osgi/bundle/org.csstudio.platform.libs.hibernate/1.0.0.201510091318/org.csstudio.platform.libs.hibernate-1.0.0.201510091318.jar>
at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:168)
at org.apache.maven.cli.MavenCli.execute(MavenCli.java:537)
at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:196)
at org.apache.maven.cli.MavenCli.main(MavenCli.java:141)
at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
at java.lang.reflect.Method.invoke(Method.java:497)
at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:290)
at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:230)
at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:414)
at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:357)
Caused by: java.lang.IllegalArgumentException: bundleLocation not found: <https://jenkins02.esss.lu.se/job/CS-Studio%20Original%2000500%20-%20cs-studio/ws/.m2/p2/osgi/bundle/org.csstudio.platform.libs.hibernate/1.0.0.201510091318/org.csstudio.platform.libs.hibernate-1.0.0.201510091318.jar>
at org.eclipse.tycho.core.osgitools.EquinoxResolver.loadManifest(EquinoxResolver.java:266)
at org.eclipse.tycho.core.osgitools.EquinoxResolver.newState(EquinoxResolver.java:161)
at org.eclipse.tycho.core.osgitools.EquinoxResolver.newResolvedState(EquinoxResolver.java:69)
at org.eclipse.tycho.core.osgitools.OsgiBundleProject.getResolverState(OsgiBundleProject.java:262)
at org.eclipse.tycho.core.osgitools.OsgiBundleProject.resolveClassPath(OsgiBundleProject.java:164)
at org.eclipse.tycho.core.resolver.DefaultTychoResolver.resolveProject(DefaultTychoResolver.java:130)
at org.eclipse.tycho.core.maven.TychoMavenLifecycleParticipant.afterProjectsRead(TychoMavenLifecycleParticipant.java:95)
at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:274)
at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:156)
... 11 more
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]

Claudio Rosati
Software Engineer

European Spallation Source ERIC
P.O Box 176, SE-221 00 Lund, Sweden
Visiting address: Tunavägen 24, Lund

Phone: +46 46 888 34 24
Mobile: +46 721 79 24 24
E-mail: claudio.rosati@esss.se<mailto:petra.ribb@esss.se>

www.esss.se<http://www.esss.se/>

[cid:EC9AC9A0-89C5-4CC1-89CE-908054AEC1B2]

kasemir, 2016-08-31:
Well, I've had several successful builds of the 'master' branch by now.

Your error truly looks like an internal maven problem:
````
[INFO] Resolving class path of MavenProject: org.csstudio:org.csstudio.platform.libs.hibernate.ui:1.0.0-SNAPSHOT @ <https://jenkins02.esss.lu.se/job/CS-Studio%20Original%2000500%20-%20cs-studio/ws/cs-studio/core/platform/platform-plugins/org.csstudio.platform.libs.hibernate.ui/pom.xml>
[ERROR] Internal error: java.lang.IllegalArgumentException: bundleLocation not found: <https://jenkins02.esss.lu.se/job/CS-Studio%20Original%2000500%20-%20cs-studio/ws/.m2/p2/osgi/bundle/org.csstudio.platform.libs.hibernate/1.0.0.201510091318/org.csstudio.platform.libs.hibernate-1.0.0.201510091318.jar> -> [Help 1]
````

In your setup, does it make sense that Maven tries to locate the org.csstudio.platform.libs.hibernate jar from an https://jenkins.. URL?
Could there be a web server issue?
Maybe need to start over after deleting $HOME/.m2/repository?

claudio-rosati, 2016-09-01:
Hello Kay,

You are right. “Copying” my local build-script into the Jenkins pipelines I forgot to clean the .m2 folder at each run.

Thank you for the suggestion. Now my pipeline works fine.

Best regards,
Claudio

Claudio Rosati
Software Engineer

European Spallation Source ERIC
P.O Box 176, SE-221 00 Lund, Sweden
Visiting address: Tunavägen 24, Lund

Phone: +46 46 888 34 24
Mobile: +46 721 79 24 24
E-mail: claudio.rosati@esss.se<mailto:petra.ribb@esss.se>

www.esss.se<http://www.esss.se/>

[cid:EC9AC9A0-89C5-4CC1-89CE-908054AEC1B2]

berryma4, 2016-09-20:
okay to close?

myhippo, 2016-09-21:
it's ok to close!
thank you!

closed 2016-09-21 by myhippo