ControlSystemStudio/cs-studio #1886: Test failure in org.csstudio.saverestore

willrogers, 2016-08-31:
```
testGetBaseLevels(org.csstudio.saverestore.git.GitManagerTest)  Time elapsed: 0.024 sec  <<< FAILURE!
java.lang.AssertionError: expected:<base2> but was:<secondBase>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:118)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at org.csstudio.saverestore.git.GitManagerTest.testGetBaseLevels(GitManagerTest.java:302)
```

It looks like the ordering in `GitManager.getBaseLevels()` is not reliable.  We've got it failing reliably over here, but it seems to be passing on Travis.

willrogers, 2016-08-31:
ttps://docs.oracle.com/javase/7/docs/api/java/io/File.html#listFiles%28%29

"There is no guarantee that the name strings in the resulting array will appear in any specific order; they are not, in particular, guaranteed to appear in alphabetical order."

nickbattam, 2016-08-31:
This is failing on both the 4.3.x and master branches

berryma4, 2016-08-31:
@jbobnar

berryma4, 2016-08-31:
Thank you @jbobnar!

commit 5a262279 (2016-08-31): Merge pull request #1887 from Cosylab/SCRBugfix

SCR Junit Test failure fix #1886
closed 2016-08-31 by berryma4