ControlSystemStudio/cs-studio #1889: Saverestore NullPointerException

jbobnar, 2016-09-01:
Nullpointer exception occurs if no specific base level browser is added to the product configuration.

java.lang.NullPointerException
	at org.csstudio.saverestore.ui.browser.BrowserView.createBaseLevelsPane(BrowserView.java:269)
	at org.csstudio.saverestore.ui.browser.BrowserView.createFxScene(BrowserView.java:244)
	at org.eclipse.fx.ui.workbench3.FXViewPart.createPartControl(FXViewPart.java:29)
	at org.csstudio.saverestore.ui.browser.BrowserView.createPartControl(BrowserView.java:199)...

jbobnar, 2016-09-01:
Can this bugfix also be applied to 4.3.x (commit https://github.com/ControlSystemStudio/cs-studio/pull/1890/commits/c73b25fb10c9d3c68e4a6f2c71bbdaebe5bf3f33)?

berryma4, 2016-09-20:
Yes, could you send a pull request to master.

jbobnar, 2016-09-20:
It has already been merged with master: https://github.com/ControlSystemStudio/cs-studio/pull/1890
Unless you meant a pull request to 4.3.x?

berryma4, 2016-09-20:
yes ... I was thinking one, and wrote the other.

berryma4, 2016-09-21:
thank you so much!!!!

closed 2016-09-21 by berryma4