ControlSystemStudio/cs-studio #1898: Test failure in org.csstudio.swt.rtplot

willrogers, 2016-09-15:
```
org.csstudio.swt.rtplot.util.UpdateThrottleTest.testUpdateThrottle

Failing for the past 1 build (Since Failed#80 )
Took 2.6 sec.
add description
Error Message

Expected: <4>
     but: was <3>

Stacktrace

java.lang.AssertionError: 
Expected: <4>
     but: was <3>
	at org.hamcrest.MatcherAssert.assertThat(MatcherAssert.java:20)
	at org.junit.Assert.assertThat(Assert.java:956)
	at org.junit.Assert.assertThat(Assert.java:923)
	at org.csstudio.swt.rtplot.util.UpdateThrottleTest.testUpdateThrottle(UpdateThrottleTest.java:76)

Standard Output

-> Trigger..
<- Update!
-> Trigger..
-> Trigger..
-> Trigger..
<- Update!
<- Update!
<- Update!
```

kasemir, 2016-09-15:
You're right, I see the same.

It is a time-based test. We hope that something has happened after about 500ms total.
But then we only actually wait 100+100+300=500ms total, which is really tight.
With 100+100+500 = 700ms I get better reproducibility.
Maybe make it 100+100+1000 to be way over on the safe side?

It could still happen that the JVM is delayed, so while we think we only waited the first 100 or 100+100=200ms, we were actually asleep for 500+ms and then you'd get an error "Expected 3 but got 4", the reverse of what you see now.

Still, I'd make that last delay 1000 instead of the current 300 and see if that works in 99% of the test runs.

kasemir, 2016-09-15:
I'll do pull request.

PR #1899 (2016-09-16): swt.rtplot: Increase time delay to allow test to pass
Fixes #1898
closed 2016-09-16 by berryma4