ControlSystemStudio/cs-studio #1908: 'Edit Embedded OPI' context menu

willrogers, 2016-09-23:
We added the option to edit an opi from run mode in BOY.  

A further user request is to be able to edit the OPI that's contained in a linking container.  This might be either in run mode or in edit mode.

kasemir, 2016-10-13:
This has also been implemented for the display builder:

In the runtime, right click anywhere to "Open in editor".
If this is done within an embedded display widget, it opens an editor for the embedded display instead of the top-level display.
If the display, embedded or top-level, is a remote http://.. link, there's a prompt for a local file name into which the display will be downloaded, and then the local copy is opened in the editor.

In the editor, the embedded display just shows the body, one cannot "touch" the widgets inside the embedded display. But when selecting the embedded display widget, the context menu "Edit Embedded Display" opens another editor for the embedded file.

Close this ticket?

willrogers, 2016-10-13:
Fine for me.

closed 2016-10-13 by willrogers