ControlSystemStudio/cs-studio #1910: Text Input Widget Malfunctions with PV Limits

dchabot, 2016-09-26:
Attn @berryma4 

Using the FRIB Debian package 4.3.2-7.

If the connecting PV for a `Text Input` widget has its `DRVH/DRVL` set, but no matching values in HOPR/LOPR, then arbitrary put-values result in the pv being set to zero.

To reproduce, simply create an `analog out` record served with `softIoc -d ai-test.db`, caput to the record to give it an initial value (say 5), caput it's DRVH to 10 (for example), and then try to use a Text Input widget to set the pv to 3. This will result in the pv being set to zero.

berryma4, 2016-09-26:
Property "Limits from PV" must be set to yes.

dchabot, 2016-09-26:
Right. Good point. Forgot that detail.

tashasummers, 2017-09-12:
Has this issue been looked into? This caused us lots of trouble again yesterday. 
The simple solution would seem to be having the "Limits from PV" property actually use the DRVH/DRVL limits instead of HOPR/LOPR. This would have to be changed for all entry widgets that take an analog value (text entry, spinner, slider).

berryma4, 2017-09-15:
@willrogers @kasemir @shroffk 
It seems for some widgets, if the property "Limits from PV" is yes and HOPR/LOPR fields are 0, then a 0 is sent to the PV.  Is this feature that people are using?  It seems the HOPR/LOPR should be for display purposes and not for setting.  Does the Display Builder have the same feature?

willrogers, 2017-09-15:
Absolutely sounds like a bug if that is what is happening.

kasemir, 2017-09-15:
In BOY, it's a bug.
"Show PV Info" shows display_low == display_high == 0, it's not getting the actual high limit.

Conceptually, I don't think the TextInput should have any min/max check.
Just like "caput", you can send any value.
The IOC may then decide to clamp it at DRVL..DRVH.
But too late for BOY, it has the min/max and "Limits from PV".
I'll see why it gets the wrong limits.

kasemir, 2017-09-15:
Looks like it's using the display limit instead of the control limit (and then falling back to the display limit in case there's no control limit).

kasemir, 2017-11-28:
Still somewhat malfunctions:
By default, ao records have DRVL==DRVH==0
--> If you enable "Limits from PV", which does sound like a good idea, you can't enter any number other than 0.

hartmansm, 2017-11-28:
The ao record only enforces limits if DRVH > DRVL. 
There are lots of records out there with DRVL==DRVH==0 that work
fine (without value clamping) with other clients. This behavior in BOY 
is unexpected.

kasemir, 2017-11-29:
TextInput now ignores the min & max unless min < max.

PR #2288 (2017-09-18): Boy Text Input: Use control instead of display range for "Limits from PV"
Fixes #1910
closed 2017-11-29 by kasemir