ControlSystemStudio/cs-studio #1911: Cannot migrate OPIs between windows anymore.

dchabot, 2016-09-26:
Why did this functionality disappear?
It's loss seems like a significant step backwards in usability/customization.
Please re-introduce this behavior, or at least make it a configuration, run-time option.

berryma4, 2016-09-26:
#1832
This is a request to be able to move a view to different workbench than it was created.

dchabot, 2016-09-26:
Related, but not quite the same thing (as I understand #1832.

berryma4, 2016-09-26:
It's the same, the view needs to be attached to another perspective.

willrogers, 2016-09-27:
As I understand it, you have two workbench windows open (i.e. using Window -> New Window to open another one) and you want to drag a view from one to the other.

This behaviour used to work in Eclipse 3 and was lost with the move to Eclipse 4.  I agree it is less usable now than it was.  However, it looks to be more-or-less impossible to fix this ourselves.

A previous ticket is #1329 - @kasemir looked at this as well.

Is this the same as you meant in #1832?

berryma4, 2016-09-27:
I clarified #1832.  Yes, I forgot about the bug, so this wouldn't be "just" moving to another perspective, but resolving https://bugs.eclipse.org/bugs/show_bug.cgi?id=384308

berryma4, 2016-09-27:
Side note.  I believe from complaints, an either or might work.
People are starting a new workbenches, at least from my discussions, so they are able to minimize the windows.
So, if views could be minimized, people wouldn't need to open the extra workbench.

berryma4, 2016-09-27:
I should also mention the view minimize behavior makes sense though.  In order to have a view minimize or have a different focus than the parent, it would need its own shell (like run as standalone does already).

dchabot, 2016-09-27:
@berryma4, are you suggesting that minimizing opi `Views` is an alternative workflow to being able to arrange `Views` across multiple `Workbench` instances?