ControlSystemStudio/cs-studio #1912: OPIs zoom to fit content, not size as defined in opi file

willrogers, 2016-09-30:
If you set the option 'zoom to fit all' on an OPI, the OPI will resize to fit the space available.

However, when zooming to fit, it takes into account all the widgets on the display, rather than the dimensions of the OPI file specified in the OPI file itself.

We find that for our OPI shells, it makes much more sense to zoom to fit the specified dimensions.  This means that you typically see what you see when the EDM screen opens, but at whatever size you have zoomed the shell to.

I have a branch to change this behaviour.  I could also change the behaviour in the workbench, but that would change the behaviour for other sites, so we should check what the consensus before doing this.

berryma4, 2016-09-30:
Is it possible to add zoom in/out hot keys to the OPIShell?

willrogers, 2016-09-30:
In this case they auto-zoom-to-fit when you change the size of the shell.  It's easy to make them zoom on ctrl-mouse-wheel but they don't have scroll bars right now so then you have to resize too.

willrogers, 2016-10-03:
Let's draw some pictures:

EDM default:
[image]

EDM 'hidden' widgets after manually resizing:

[image]

OPIShell - _not_ zoom to fit all
[image]

OPIShell - zoom to fit all

[image]

Unfortunately, as I write this, the current behaviour does _seem_ to make sense.  It causes us a problem because we want to make our OPIShells auto-resize when the user resizes the window, and what the user sees in EDM (and therefore, what we think they should be seeing) is not necessarily the same as all the widgets contained in the screen.

@kasemir are there any lessons in display builder?  Can you auto-resize your screens?  N.b. in Boy the screen scaling/resizing/zooming does work very well.

kasemir, 2016-10-20:
BOY does the same for embedded displays (linking container):
When the body is resized to fit the container, that's done based on the actual size of the body widgets, not based on the width, height of the embedded display file.

> We find .. it makes much more sense to zoom to fit the specified dimensions.
That's what I thought, too, so the display builder scales the body of embedded displays based on the self-declared dimensions of the embedded display.
Still, any scaling usually results in fonts being different sizes than the parent display, it just looks odd, so the default is to not resize, getting scrollbars if it doesn't fit.

For the overall "zoom to width", "..height" or "..all", the display builder _does_ use the actual size of all widgets in the display. The specified dimensions of the display are only used when opening a new display in the standalone runtime.

Reason for using the specified size when scaling embedded content vs. using the actual visual size when zooming interactively via the runtime's toolbar:
The embedded display configuration is a design-time decision where a knowledgeable person has to configure the scaling mode and dimensions of the display in the body. When that same knowledgeable person later moves some widget around in the embedded display, that should not suddenly change the scaling.
Zooming in the display runtime on the other hand is done by an end user. When they want to "zoom all", they want to see all, based on the visible widget. They don't care what the specified size might be.

willrogers, 2017-01-24:
Done.

closed 2017-01-24 by willrogers