ControlSystemStudio/cs-studio #1919: cd cs-studio/applications; mvn clean verify ,then there are some errors.why?how to solve it?

ksspwuwei, 2016-10-11:
[ERROR] An error occurred while transferring artifact canonical: osgi.bundle,org.csstudio.platform.libs.jms,5.4.2.201510091318 from repository http://download.controlsystemstudio.org/thirdparty/4.4:
[ERROR]    Unable to read repository at http://download.controlsystemstudio.org/thirdparty/4.4/plugins/org.csstudio.platform.libs.jms_5.4.2.201510091318.jar.
[ERROR] Internal error: org.eclipse.tycho.repository.local.MirroringArtifactProvider$MirroringFailedException: Could not mirror artifact osgi.bundle,org.csstudio.platform.libs.jms,5.4.2.201510091318 into the local Maven repository.See log output for details. Read timed out -> [Help 1]
org.apache.maven.InternalErrorException: Internal error: org.eclipse.tycho.repository.local.MirroringArtifactProvider$MirroringFailedException: Could not mirror artifact osgi.bundle,org.csstudio.platform.libs.jms,5.4.2.201510091318 into the local Maven repository.See log output for details.
	at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:121)
	at org.apache.maven.cli.MavenCli.execute(MavenCli.java:863)
	at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:288)
	at org.apache.maven.cli.MavenCli.main(MavenCli.java:199)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:289)
	at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:229)
	at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:415)
	at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:356)
Caused by: org.eclipse.tycho.repository.local.MirroringArtifactProvider$MirroringFailedException: Could not mirror artifact osgi.bundle,org.csstudio.platform.libs.jms,5.4.2.201510091318 into the local Maven repository.See log output for details.
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.downloadArtifact(MirroringArtifactProvider.java:226)
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.makeOneFormatLocallyAvailable(MirroringArtifactProvider.java:211)
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.makeLocallyAvailable(MirroringArtifactProvider.java:182)
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.getArtifactFile(MirroringArtifactProvider.java:121)
	at org.eclipse.tycho.repository.p2base.artifact.provider.CompositeArtifactProvider.getArtifactFile(CompositeArtifactProvider.java:85)
	at org.eclipse.tycho.p2.target.TargetPlatformBaseImpl.getLocalArtifactFile(TargetPlatformBaseImpl.java:128)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.addUnit(P2ResolverImpl.java:254)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.toResolutionResult(P2ResolverImpl.java:217)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.resolveDependencies(P2ResolverImpl.java:209)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.resolveDependencies(P2ResolverImpl.java:116)
	at org.eclipse.tycho.p2.resolver.P2DependencyResolver.doResolveDependencies(P2DependencyResolver.java:367)
	at org.eclipse.tycho.p2.resolver.P2DependencyResolver.resolveDependencies(P2DependencyResolver.java:335)
	at org.eclipse.tycho.core.resolver.DefaultTychoResolver.resolveProject(DefaultTychoResolver.java:117)
	at org.eclipse.tycho.core.maven.TychoMavenLifecycleParticipant.afterProjectsRead(TychoMavenLifecycleParticipant.java:95)
	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:266)
	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:193)
	at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:106)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:170)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at org.apache.http.impl.io.AbstractSessionInputBuffer.read(AbstractSessionInputBuffer.java:198)
	at org.apache.http.impl.io.ContentLengthInputStream.read(ContentLengthInputStream.java:178)
	at org.apache.http.conn.EofSensorInputStream.read(EofSensorInputStream.java:137)
	at java.io.FilterInputStream.read(FilterInputStream.java:133)
	at java.io.FilterInputStream.read(FilterInputStream.java:107)
	at org.eclipse.ecf.provider.filetransfer.retrieve.AbstractRetrieveFileTransfer$1.performFileTransfer(AbstractRetrieveFileTransfer.java:178)
	at org.eclipse.ecf.filetransfer.FileTransferJob.run(FileTransferJob.java:73)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/InternalErrorException

ksspwuwei, 2016-10-11:
if "mvn clean compile",then there are some errors:

 [ERROR] An error occurred while transferring artifact canonical: osgi.bundle,com.google.guava,18.0.0 from repository http://download.controlsystemstudio.org/maven-osgi-bundles/4.4:
[ERROR]    Unable to read repository at http://download.controlsystemstudio.org/maven-osgi-bundles/4.4/plugins/com.google.guava_18.0.0.jar.
[ERROR] Internal error: org.eclipse.tycho.repository.local.MirroringArtifactProvider$MirroringFailedException: Could not mirror artifact osgi.bundle,com.google.guava,18.0.0 into the local Maven repository.See log output for details. Read timed out -> [Help 1]
org.apache.maven.InternalErrorException: Internal error: org.eclipse.tycho.repository.local.MirroringArtifactProvider$MirroringFailedException: Could not mirror artifact osgi.bundle,com.google.guava,18.0.0 into the local Maven repository.See log output for details.
	at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:121)
	at org.apache.maven.cli.MavenCli.execute(MavenCli.java:863)
	at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:288)
	at org.apache.maven.cli.MavenCli.main(MavenCli.java:199)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:289)
	at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:229)
	at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:415)
	at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:356)
Caused by: org.eclipse.tycho.repository.local.MirroringArtifactProvider$MirroringFailedException: Could not mirror artifact osgi.bundle,com.google.guava,18.0.0 into the local Maven repository.See log output for details.
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.downloadArtifact(MirroringArtifactProvider.java:226)
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.makeOneFormatLocallyAvailable(MirroringArtifactProvider.java:211)
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.makeLocallyAvailable(MirroringArtifactProvider.java:182)
	at org.eclipse.tycho.repository.local.MirroringArtifactProvider.getArtifactFile(MirroringArtifactProvider.java:121)
	at org.eclipse.tycho.repository.p2base.artifact.provider.CompositeArtifactProvider.getArtifactFile(CompositeArtifactProvider.java:85)
	at org.eclipse.tycho.p2.target.TargetPlatformBaseImpl.getLocalArtifactFile(TargetPlatformBaseImpl.java:128)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.addUnit(P2ResolverImpl.java:254)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.toResolutionResult(P2ResolverImpl.java:217)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.resolveDependencies(P2ResolverImpl.java:209)
	at org.eclipse.tycho.p2.resolver.P2ResolverImpl.resolveDependencies(P2ResolverImpl.java:116)
	at org.eclipse.tycho.p2.resolver.P2DependencyResolver.doResolveDependencies(P2DependencyResolver.java:367)
	at org.eclipse.tycho.p2.resolver.P2DependencyResolver.resolveDependencies(P2DependencyResolver.java:335)
	at org.eclipse.tycho.core.resolver.DefaultTychoResolver.resolveProject(DefaultTychoResolver.java:117)
	at org.eclipse.tycho.core.maven.TychoMavenLifecycleParticipant.afterProjectsRead(TychoMavenLifecycleParticipant.java:95)
	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:266)
	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:193)
	at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:106)
	... 11 more
Caused by: java.net.SocketTimeoutException: Read timed out
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:170)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at org.apache.http.impl.io.AbstractSessionInputBuffer.read(AbstractSessionInputBuffer.java:198)
	at org.apache.http.impl.io.ContentLengthInputStream.read(ContentLengthInputStream.java:178)
	at org.apache.http.conn.EofSensorInputStream.read(EofSensorInputStream.java:137)
	at java.io.FilterInputStream.read(FilterInputStream.java:133)
	at java.io.FilterInputStream.read(FilterInputStream.java:107)
	at org.eclipse.ecf.provider.filetransfer.retrieve.AbstractRetrieveFileTransfer$1.performFileTransfer(AbstractRetrieveFileTransfer.java:178)
	at org.eclipse.ecf.filetransfer.FileTransferJob.run(FileTransferJob.java:73)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/InternalErrorException

berryma4, 2016-10-11:
```
Caused by: java.net.SocketTimeoutException: Read timed out
```
I'm going to assume S3 (our web server) was momentarily down.
Please try again.

ksspwuwei, 2016-10-12:
@berryma4 
## Now,I "mvn clean verify",there are some new errors,how to solve it?thank you for your help!

[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.alarm.beast 4.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.alarm.beast 4.2.0.qualifier requires 'bundle org.csstudio.csdata 3.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException
[root@wuwei applications]# ^C
[root@wuwei applications]# mvn clean compile
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/releases/mars
[INFO] Fetching p2.index from http://download.eclipse.org/technology/epp/packages/mars/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201506241002/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201510021000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201602261000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/eclipse/updates/4.5
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5-201506032000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.1-201509040015/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.2-201602121500/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/rt/rap/3.0
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/R-20150602-0950/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR1a-20150908-1130/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR2-20160201-1220/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/rt/rap/incubator/3.0/e4/target/site
[INFO] Adding repository http://download.eclipse.org/rt/rap/incubator/nightly/gef/20150327-1849
[INFO] Adding repository http://download.eclipse.org/tools/orbit/downloads/drops/R20160221192158/repository
[INFO] Adding repository http://download.eclipse.org/efxclipse/runtime-released/2.1.0/site
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.alarm.beast 4.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.alarm.beast 4.2.0.qualifier requires 'bundle org.csstudio.csdata 3.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException
[root@wuwei applications]# ^C
[root@wuwei applications]# ^C
[root@wuwei applications]# mvn clean verify
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/releases/mars
[INFO] Fetching p2.index from http://download.eclipse.org/technology/epp/packages/mars/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201506241002/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201510021000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201602261000/ (0B at 0B/s)
[... cut ...]
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

ksspwuwei, 2016-10-12:
## @berryma4 
Now,I met some new errors,how to solve it?Thank you for your help!

[root@wuwei applications]# mvn clean verify
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/releases/mars
[INFO] Fetching p2.index from http://download.eclipse.org/technology/epp/packages/mars/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201506241002/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201510021000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201602261000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/eclipse/updates/4.5
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5-201506032000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.1-201509040015/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.2-201602121500/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/rt/rap/3.0
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/R-20150602-0950/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR1a-20150908-1130/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR2-20160201-1220/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/rt/rap/incubator/3.0/e4/target/site
[INFO] Adding repository http://download.eclipse.org/rt/rap/incubator/nightly/gef/20150327-1849
[INFO] Adding repository http://download.eclipse.org/tools/orbit/downloads/drops/R20160221192158/repository
[INFO] Adding repository http://download.eclipse.org/efxclipse/runtime-released/2.1.0/site
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.alarm.beast 4.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.alarm.beast 4.2.0.qualifier requires 'bundle org.csstudio.csdata 3.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

berryma4, 2016-10-12:
Do you have a settings.xml file like stated in the documentation:
http://cs-studio.sourceforge.net/docbook/ch04.html

kasemir, 2016-10-12:
Hi:

> On Oct 12, 2016, at 8:06 AM, Eric Berryman <notifications@github.com> wrote:
> Do you have a settings.xml file like stated in the documentation:
> http://cs-studio.sourceforge.net/docbook/ch04.html
> 
You can also compare with the build setup used here:
https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build

The SNS product’s build/ directory contains a get_all.sh script that fetches all the required git repos, and then there’s a make.sh that compiles them all, invoking mvn with command-line settings to point to the org.csstudio.sns/build/settings.xml, which in turn references the composite org.csstudio.sns/css_repo.

That might help to clarify what’s described in more detail in the documentation.

-Kay

ksspwuwei, 2016-10-13:
@berryma4 ## where should I put the settings.xml?
## I creat the settings.xml in the /root/.m2/,then: 
[root@wuwei maven-osgi-bundles-master]# mvn help:active-profiles
[INFO] Scanning for projects...
[WARNING] 
[WARNING] Some problems were encountered while building the effective model for org.csstudio:org.csstudio.platform.libs.epics:bundle:4.3.0-SNAPSHOT
[WARNING] 'version' contains an expression but should be a constant. @ org.csstudio:org.csstudio.platform.libs.epics:${csstudio.version}, /root/cs-studio/maven-osgi-bundles-master/epics/plugins/org.csstudio.platform.libs.epics/pom.xml, line 13, column 12
[WARNING] 'build.plugins.plugin.(groupId:artifactId)' must be unique but found duplicate declaration of plugin org.codehaus.mojo:build-helper-maven-plugin @ org.csstudio:org.csstudio.platform.libs.epics:${csstudio.version}, /root/cs-studio/maven-osgi-bundles-master/epics/plugins/org.csstudio.platform.libs.epics/pom.xml, line 82, column 15
[WARNING] 
[WARNING] It is highly recommended to fix these problems because they threaten the stability of your build.
[WARNING] 
[WARNING] For this reason, future Maven versions might no longer support building such malformed projects.
[WARNING] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] maven-osgi-bundles
[INFO] epics
[INFO] epics-plugins
[INFO] org.csstudio.platform.libs.epics
[INFO] maven-osgi-repository
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building maven-osgi-bundles 1.0.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-help-plugin:2.2:active-profiles (default-cli) @ maven-osgi-bundles ---
[INFO] 
Active Profiles for Project 'org.csstudio:maven-osgi-bundles:pom:1.0.0-SNAPSHOT': 

The following profiles are active:

 - my-css-settings (source: external)

Active Profiles for Project 'org.csstudio:epics:pom:0.0.1-SNAPSHOT': 

The following profiles are active:

 - my-css-settings (source: external)

Active Profiles for Project 'org.csstudio:epics-plugins:pom:0.0.1-SNAPSHOT': 

The following profiles are active:

 - my-css-settings (source: external)

Active Profiles for Project 'org.csstudio:org.csstudio.platform.libs.epics:bundle:4.3.0-SNAPSHOT': 

The following profiles are active:

 - my-css-settings (source: external)

Active Profiles for Project 'org.csstudio:maven-osgi-repository:jar:1.0.0-SNAPSHOT': 

The following profiles are active:

 - my-css-settings (source: external)

[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] maven-osgi-bundles ................................. SUCCESS [  0.408 s]
[INFO] epics .............................................. SKIPPED
[INFO] epics-plugins ...................................... SKIPPED
[INFO] org.csstudio.platform.libs.epics ................... SKIPPED
[INFO] maven-osgi-repository .............................. SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 1.786 s
[... 52 lines cut ...]
The following profiles are active:

 - my-css-settings (source: external)

Active Profiles for Project 'org.csstudio:maven-osgi-repository:jar:1.0.0-SNAPSHOT': 

The following profiles are active:

 - my-css-settings (source: external)

[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] maven-osgi-bundles ................................. SUCCESS [  0.288 s]
[INFO] epics .............................................. SKIPPED
[INFO] epics-plugins ...................................... SKIPPED
[INFO] org.csstudio.platform.libs.epics ................... SKIPPED
[INFO] maven-osgi-repository .............................. SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 1.656 s
[INFO] Finished at: 2016-10-13T10:55:07+08:00
[INFO] Final Memory: 16M/169M
[INFO] ------------------------------------------------------------------------

berryma4, 2016-10-13:
you should put the settings.xml file in your ~/.m2 directory

kasemir, 2016-10-13:
It depends. Your ~/.m2/settings.xml are 'global', they will apply whenever you call maven.
I tend to call maven as
````
mvn -s /path/to/my/settings.xml
````
because I prefer to know exactly which settings file it uses, without depending on some file being in a magic location. Also makes it easier to change between
````
mvn -s /path/to/my/settings1.xml
````
and
````
mvn -s /path/to/my/settings2.xml
````

See https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build for the SNS build setup.

ksspwuwei, 2016-10-14:
@berryma4 @kasemir ,## Thank you  for your help!
I have create the settings.xml file,and put it in ~/.m2 directory.
## But after I cd core;mvn -B -P my-css-settings clean verify;then:
[root@wuwei cs-studio-master]# cd core/
[root@wuwei core]# mvn -B -P my-css-settings clean verify
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/core/utility/utility-plugins/org.csstudio.data/pom.xml
[INFO] Adding repository file:/root/cs-studio/my_comp_repo
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/core/utility/utility-plugins/org.csstudio.data/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   You requested to install 'bundle javax.servlet 0.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/core/utility/utility-plugins/org.csstudio.data/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

## And if I cd applications;mvn -B -P my-css-settings clean verify,then:
[root@wuwei cs-studio-master]# cd applications/
[root@wuwei applications]# mvn -B -P my-css-settings clean verify
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml
[INFO] Adding repository file:/root/cs-studio/my_comp_repo
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/releases/mars
[INFO] Fetching p2.index from http://download.eclipse.org/technology/epp/packages/mars/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201506241002/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201510021000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201602261000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/eclipse/updates/4.5
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5-201506032000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.1-201509040015/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.2-201602121500/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/rt/rap/3.0
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/R-20150602-0950/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR1a-20150908-1130/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR2-20160201-1220/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/rt/rap/incubator/3.0/e4/target/site
[INFO] Adding repository http://download.eclipse.org/rt/rap/incubator/nightly/gef/20150327-1849
[INFO] Adding repository http://download.eclipse.org/tools/orbit/downloads/drops/R20160221192158/repository
[INFO] Adding repository http://download.eclipse.org/efxclipse/runtime-released/2.1.0/site
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.alarm.beast 4.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.alarm.beast 4.2.0.qualifier requires 'bundle org.csstudio.csdata 3.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.2.0-SNAPSHOT @ /root/cs-studio/cs-studio-master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

## I'm very sorry to disturb you,what I can say is thank you very much!

kasemir, 2016-10-14:
You're building cs-studio/applications, and it can't find org.csstudio.csdata, which is in cs-studio/core.

Did you build cs-studio/code?
Successfully?
Do your my-css-settings define a csstudio.composite.repo which somehow lists either your locally built ../../cs-studio/core/p2repo (that's what I prefer), or do you list some web repository (then you're on your own with trying to debug that)?

Compare your setup to https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build:
My settings.xml sets csstudio.composite.repo to ${env.SRC_DIR}/org.csstudio.sns/css_repo, i.e. a copy of https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/css_repo.

In there, check the *.xml files which both list ../../cs-studio/core/p2repo, i.e. the output of the cs-studio/core build that I perform before trying to build cs-studio/applications.
In fact the make.sh in https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build shows the exact steps how I build CSS each night, so try that.

lotrus, 2016-11-04:
Hi,
We are working on compiling CSS from source too based on http://cs-studio.sourceforge.net/docbook/ch04.html explanation, and we have the same error on building cs-studio/core:
C:\css_sources_manual\cs-studio\core>mvn -B -P my-css-settings clean verify
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ C:\css_sources_manual\cs-studio\core\utility\utili
y-plugins\org.csstudio.data\pom.xml
[INFO] Adding repository file:/C:/css_sources_manual/my_comp_repo
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ C:\css_sources_manual\cs-studio\core\utility\utility-p
ugins\org.csstudio.data\pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   You requested to install 'bundle javax.servlet 0.0.0' but it could not be found
[ERROR]
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.data:3.2.0-SNAPSHOT @ C:\css_sources_manual\cs-studio\core\utility\uti
ity-plugins\org.csstudio.data\pom.xml: See log for details -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

Thyco website (http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting) provides this information:
_You requested to install '<reference>' but it could not be found
This message is shown when one of the extraRequirements specified in the target-platform-configuration cannot be resolved. The <reference> is printed in the same format as the dependencies in missing requirements messages (see below for details)._

Something should be missing in our configuration about the target Platform, but we don't find it.

The cs-studio/core/pom.xml file contains a requirement on javax.servlet but for some reason.
```xml
…
  <build>
    <plugins>
      <plugin>
        <groupId>org.eclipse.tycho</groupId>
        <artifactId>target-platform-configuration</artifactId>
        <version>${tycho.version}</version>
        <configuration>
          <dependency-resolution>
            <extraRequirements>
…
             <requirement>
                <type>eclipse-plugin</type>
                <id>javax.servlet</id>
                <versionRange>0.0.0</versionRange>
              </requirement>
```
Our settings.xml is:
```xml
<?xml version="1.0" encoding="UTF-8"?>
 <settings xmlns="http://maven.apache.org/SETTINGS/1.0.0"
          xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
          xsi:schemaLocation="http://maven.apache.org/SETTINGS/1.0.0 http://maven.apache.org/xsd/settings-1.0.0.xsd">
  <localRepository>C:/repositories/maven</localRepository>
  <mirrors>
     <mirror> 
       <id>maven-central</id> 
       <name>Maven Central</name> 
       <url>https://repo1.maven.org/maven2</url> 
       <mirrorOf>central</mirrorOf> 
    </mirror> 
    <mirror> 
       <id>google-maven-central</id> 
       <name>Google Maven Central</name> 
       <url>https://maven-central.storage.googleapis.com</url> 
       <mirrorOf>central</mirrorOf> 
     </mirror> 
  </mirrors>
  <profiles>
    <profile>
      <id>my-css-settings</id>
      <properties>        <csstudio.composite.repo>/C:/css_sources_manual/my_comp_repo</csstudio.composite.repo>
        <cs-studio>false</cs-studio>
        <eclipse>false</eclipse>
        <tycho.localArtifacts>ignore</tycho.localArtifacts>
        <baselineMode>warn</baselineMode>
        <jgit.dirtyWorkingTree>warning</jgit.dirtyWorkingTree>
        <maven.test.skip>true</maven.test.skip>
        <skipTests>true</skipTests>
      </properties>
    </profile>
  </profiles>
  <activeProfiles>
    <activeProfile>my-css-settings</activeProfile>
  </activeProfiles>
</settings>
```

Our Maven configuration is:
Apache Maven 3.3.9 (bb52d8502b132ec0a5a3f4c09453c07478323dc5; 2015-11-10T17:41:47+01:00)
Maven home: C:\bin\apache-maven-3.3.9
Java version: 1.8.0_101, vendor: Oracle Corporation
Java home: C:\Program Files\Java\jdk1.8.0_101\jre
Default locale: fr_FR, platform encoding: Cp1252
OS name: "windows 7", version: "6.1", arch: "amd64", family: "dos"

Does anyone have a idea?

Regards

kasemir, 2016-11-04:
No idea. I'm mostly happy that it works for me, most of the time, and the best I can do is continue to point to the exact setup that I'm using: https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build

I'm unclear why it tries to "install" anything.
I just use "mvn clean verify".
Why are you using "mvn -B" with the "-B" option? Does that somehow ask it to "install"?

lotrus, 2016-11-04:
Thanks, we could go farther by changing our settings.xml by removing following parameters:
```xml
        <cs-studio>false</cs-studio>
        <eclipse>false</eclipse>
```
But we have another build failure:
```
[INFO] org.csstudio.core.diirt.feature .................... SUCCESS [  0.749 s]
[INFO] diirt-repository ................................... FAILURE [  0.702 s]
[INFO] org.csstudio.vtype.pv.test ......................... SKIPPED
...
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 02:32 min
[INFO] Finished at: 2016-11-04T17:11:27+01:00
[INFO] Final Memory: 351M/1287M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-p2-repository-plugin:0.23.1:assemble-repository (default-assemble-repository) on project diirt-
repository: Execution default-assemble-repository of goal org.eclipse.tycho:tycho-p2-repository-plugin:0.23.1:assemble-repository failed: Cannot resol
ve dependencies of MavenProject: org.csstudio:diirt-repository:4.2.0-SNAPSHOT @ C:\css_sources_manual\cs-studio\core\diirt\diirt-repository\pom.xml: S
ee log for details -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginExecutionException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :diirt-repository
```

kasemir, 2016-11-04:
Could indicate that your csstudio.composite.repo isn't functional.

Defining the property `cs-studio` _disables_ access to `http://diirt.download.controlsystemstudio.org/diirt`, `${download.root}/thirdparty`, `${download.root}/maven-osgi-bundles` etc., so it's then mostly using your own composite repo. See https://github.com/ControlSystemStudio/cs-studio/blob/master/core/pom.xml#L105

You have configured a composite repo, it's trying to use 'my_comp_repo':
````
 [INFO] Adding repository file:/C:/css_sources_manual/my_comp_repo
````
.. but maybe it points to things that don't actually exist, and because  you have set  `p2.atomic.composite.loading` to `false`, there are no errors for missing child locations of the composite repo?

kasemir, 2016-11-04:
The `eclipse` property should indeed not be defined, so that it does go out to the Eclipse download sites.
So overall, this should work:
````
<!-- Define cs-studio to _disable_ remote downloads for CSS
     bundles, instead using them from locally created composite repo -->
<cs-studio>any_dummy_value</cs-studio>
<!-- Do NOT define eclipse, so it does download from Eclipse repos -->
<!--eclipse>any_dummy_value</eclipse-->
````

lotrus, 2016-11-07:
Thanks, but we have now an error when compiling core about a dependency on sun.misc:
```
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.core.diirt.feature:0.0.1-SNAPSHOT @ C:\css_sources_manual\cs-studio\core\diirt\diirt-features\org.csstudio.core.diirt.feature\pom.xml
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.core.diirt.feature:0.0.1-SNAPSHOT @ C:\css_sources_manual\cs-studio\core\diirt\diirt-features\org.csstudio.core.diirt.feature\pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.core.diirt.feature.feature.group 0.0.1.qualifier
[ERROR]   Missing requirement: org.epics.pvAccessJava 4.1.3 requires 'package sun.misc 0.0.0' but it could not be found
[ERROR]   Cannot satisfy dependency: org.csstudio.core.diirt.feature.feature.group 0.0.1.qualifier depends on: org.diirt.support.diirt-pva 0.0.0
[ERROR]   Cannot satisfy dependency: org.diirt.support.diirt-pva 3.1.5 depends on: package rg.epics.pvaccess [4.1.0,5.0.0)
[ERROR]
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.core.diirt.feature:0.0.1-SNAPSHOT @ C:\css_sources_manual\cs-studio\core\diirt\diirt-features\org.csstudio.core.diirt.feature\pom.xml: See log for details -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException
```
We have read that there are issues about it, but we are a little bit lost on how to deal with it. We are compiling the branch 4.3.x. Any suggestions for us? Are we on the wrong branch of the tree?

kasemir, 2016-11-07:
The dependency on sun.misc has been fixed and the o.epics.pvaccess versions have changed. It's OK on the master branch. Unfortunately the times where you could just get a zip with all the sources for an older version are long gone. I have no idea what you would have to do to still get it to build on an older branch.

berryma4, 2016-11-07:
If you are building diirt from sources, instead of pulling the binaries over p2 (from http://controlsystemstudio/diirt), you will need to check out the correct version of diirt associated with the cs-studio branch (the version specified in the core/pom.xml as the diirt.version property).

lotrus, 2016-11-10:
Thanks for your answers, we are able to compile with Maven. We had made two errors: 
- not working on master 
- bad path in compositeArtifact.xml and compositeContent.xml located in the csstudio.composite.repo

ksspwuwei, 2016-11-10:
@lotrus ,can you tell me that how did solve the second error?Before you compile the cs-studio/core,have you compile the cs-studio-thirdparty successfully?I compile the  cs-studio-thirdparty unsuccessfully.

ksspwuwei, 2016-11-14:
I have compiled the cs-studio/core successfully,but when I compile the cs-studio/applications,there are errors:
[ERROR] GC overhead limit exceeded -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/OutOfMemoryError

Can anyone tell me how to solve it?

lotrus, 2016-11-14:
Java memory size by default is not big enough for cs-sutdio/applications, we created a MAVEN_OPTS environment variable like this to set the size at 2Gb:
set MAVEN_OPTS=-Xms512m –Xmx2048m

Sorry, I have not received a notification for the first question, the second error was solved by working on the last master git with the command: git checkout master

ksspwuwei, 2016-11-23:
@lotrus Thank you for your answer!I input the order:
export MAVEN_OPTS='-Xms512m -Xmx4096m -XX:-UseGCOverheadLimit -XX:+UseConcMarkSweepGC'
And then I compiled servral times,finally I compiled it successfully!

closed 2020-07-23 by shroffk