ControlSystemStudio/cs-studio #1928: Better error message from BOY or, avoiding this kind of messages on Display Builder

claudio-rosati, 2016-10-18:
Hello all,

my colleague at ESS is using an embedded BOY OPI into a master one. Running this master one, the following error appeared:

```
2016-10-18 15:04:11 ERROR: Error in  on LNS-LEBT-010:VAC-VPDP-00031.
org.mozilla.javascript.EcmaError: TypeError: Body is not an xml object. (rule#65)
```

That embedded OPI does't use any rule at all, but looking at the OPI itself I've found the following:

```
...
<scriptText><![CDATA[importPackage(Packages.org.csstudio.opibuilder.scriptUtil);

//This script will perform all the animations of this window

//Get the PVs
var pvManMode = PVUtil.getDouble(pvs[0]);
var pvStopped = PVUtil.getDouble(pvs[1]);
var pvRunOutput = PVUtil.getDouble(pvs[2]);
var pvLocalMode = PVUtil.getDouble(pvs[3]);
...
else if(pvStopped==0 && pvRunOutput==1 && pvNominalSpeed == 0)
	wgBody.setPropertyValue("image_file","../images/vac-vpdp-starting.png");
else if(pvStopped==0 && pvRunOutput==1 && pvNominalSpeed == 1)
	wgBody.("image_file","../images/vac-vpdp-on.png");            <<<===
else
	wgBody.setPropertyValue("image_file","../images/vac-vpdp-off.png");
...
```

The error was a missing `setPropertyValue` on the line marked by `<<<===`, line that happen to be the n. 65 of the script, starting at  `<scriptText><![CDATA[importPackage...`.

So my request is the following: when there is an error in the script, can the error message end with `(script line#XXX)` instead of `(rule#XXX)`?

Thank you,
Claudio

kasemir, 2016-11-11:
Depends.

If I mistype the `setPropertyValue`, the error messages in the display builder runtime looks like this, with a nice line number in the error message:
````
2016-11-11 12:41:03.158 WARNING [Thread 71] org.csstudio.display.builder.runtime.script.internal.JythonScriptSupport (lambda$0) - Execution of 'JythonScript EmbeddedPy' failed for Widget 'Label_28' (label)
Traceback (most recent call last):
  File "EmbeddedPy", line 6, in <module>
AttributeError: 'org.csstudio.display.builder.model.widgets.LabelWi' object has no attribute 'setPropertyValueXXXX'
````

On the other hand, if I remove the `setPropertyValue` as in your example, the error message becomes
````
2016-11-11 12:41:27.342 WARNING [Thread 81] org.csstudio.display.builder.runtime.WidgetRuntime (startScripts) - Widget Label_28 script EmbeddedPy failed to initialize
SyntaxError: ("mismatched input '(' expecting set null", ('EmbeddedPy', 6, 7, 'widget.("text", path)\n'))
````

So either way we see the error reported by Jython plus the stack trace inside the JVM that leads into the Jython parser, which I think is better than the BOY error message.

claudio-rosati, 2016-11-14:
Yes, I agree.
I don't know if BOY errors can be improved, or if is not worth being Display Builder getting ready very quickly ;-)