ControlSystemStudio/cs-studio #1937: Print action from opi runtime results in cs-studio crash

shroffk, 2016-10-24:
```
The program 'Cs-studio' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadDrawable (invalid Pixmap or Window parameter)'.
  (Details: serial 42818 error_code 9 request_code 53 minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

nickbattam, 2016-10-24:
@shroffk Which OS/window manager(s) are you seeing this problem with?

I can print from an opi open in the OPI Runtime (right-click on opi) on my RHEL6 / GTK+2 machine using both the 4.3 and master branches with no error.

berryma4, 2016-10-24:
I can reproduce this on the 4.3.x branch, linux, gnome desktop.

shroffk, 2016-10-24:
Debian 8 with Gnome and KDE
with export SWT_GTK3=0 
So hopefully this should mean that we are still using gtk2

kasemir, 2016-10-24:
My setup is similar to Nick: RHEL 6.8 , and Help/About/../Configuration shows
`org.eclipse.swt.internal.gtk.version=2.24.23`.

Any more detail on the crash? Is there a stack trace that includes print-related methods which then end in some GTK/X call, and that receives the BadDrawable error?

shroffk, 2016-10-24:
There isn't a nice sack trace to look at, the abrupt exit from the jvm sort of makes it difficult to capture some of this information....

org.eclipse.swt.internal.gtk.cairoGraphics=false
org.eclipse.swt.internal.gtk.useCairo=false
org.eclipse.swt.internal.gtk.version=2.24.10
 
Later this week I will be able to provide more details..

berryma4, 2016-10-25:
I'm gtk version 2.24.25.
I can't seem to get a backtrace on the error.
The gdk_x_error break point never gets hit.
using:
```
GTK_DEBUG="all" GDK_SYNCHRONIZE="1" gdb /usr/lib/cs-studio/cs-studio

(gdb) break exit
(gdb) break gdk_x_error
(gdb) catch throw
(gdb) r -debug -pluginCustomization=debug.ini
```
I get a break at exit, but nothing for the gdk_x_error.
Any ideas so I can see the problem?
I can't seem to get the actual error.

berryma4, 2016-10-25:
... funny, gtk printing is owned by Matthias Clasen, who works for Red Hat ....

berryma4, 2016-12-21:
Okay, I setup a remote debugger, then stepped through PrintDisplayAction.java.
I can't reproduce the issue in a debugger.  It works fine.  so:

works fine with or without breakpoints
```
cs-studio -debug -vmargs -Xdebug -Xrunjdwp:transport=dt_socket,suspend=y,server=y,address=4444
```

doesn't work:
```
cs-studio -debug
```

doesn't work:
```
cs-studio
```

berryma4, 2016-12-21:
@shroffk I know you also have this issue (also use debian).
Any ideas?  Threading issue?

berryma4, 2016-12-21:
Maybe there is an issue with this asyncExec in the runnable.

berryma4, 2016-12-22:
@shroffk I made a pull request you can try.  I basically just move somethings around, but I did lengthen the catch.  Hoping we can see the error.

berryma4, 2016-12-22:
Okay, this didn't help, I still don't get a stacktrace.

berryma4, 2017-01-03:
``` java
printerImage.dispose();
```
is that line that kills it in PrintDisplayAction

There was a similar bug for OSX a while back.  I used the same work around, to move the dispose after printing is done.

https://bugs.eclipse.org/bugs/show_bug.cgi?id=282998

This seems to fix the issue.  Although I don't care for the fix (seems a bit hacky).
@shroffk  Does it work for you?
If so, I'll merge on to 4.3.x and master.

berryma4, 2017-01-03:
@kasemir @nickbattam I will probably need a RedHat person to also verify that I fixed Debian, yet didn't kill RedHat.

kasemir, 2017-01-03:
Looks OK to me. I can test when it's on master.

kasemir, 2017-01-04:
Thanks for tracking this down!
Does it look like the main issue was 
````
# Disposing the image that's being printed too early
printerImage.dispose();
printer.endPage();
printer.endJob();
printer.dispose();
````

````
printer.dispose();
# OK to dispose image once the printer is gone
printerImage.dispose();
````
?

I'll do that in the display builder then.

berryma4, 2017-01-04:
Yes, I'm guessing that was the issue ... and why in debug mode the crash didn't happen.

berryma4, 2017-01-04:
@shroffk I will close the ticket if you can confirm.

closed 2017-01-23 by berryma4