ControlSystemStudio/cs-studio #1938: caj kills IOC

berryma4, 2016-10-24:
I've had two cases in which someone was restarting their IOC a few times, then the IOC gets bogged down with invalid request:
```
CAS: forcing disconnect from 192.168.xxx.xxx:42365
CAS: request from 192.168.xxx.xxx:42367 => "bad resource ID"
CAS: Request from 192.168.xxx.xxx:42367 => cmmd=2 cid=0x11 type=20 count=1 postsize=0
CAS: Request from 192.168.xxx.xxx:42367 => available=0x3c974 N=0 paddr=0x36fbe08
CAS: forcing disconnect from 192.168.xxx.xxx:42367
CAS: forcing disconnect from 192.168.xxx.xxx:42370
```
This sounds like our previous issue, but showing itself in a different way.
https://github.com/ControlSystemStudio/cs-studio/issues/1174

kasemir, 2016-10-24:
Is this with BOY where you could try both pvmanager and vtype_pv?
Maybe even another CAJ-based client (well, not sure which one), so we see if this is in CAJ or the result of using CAJ in a certain way.

berryma4, 2016-10-24:
I will try to reproduce with both.