ControlSystemStudio/cs-studio #1939: diirt archive datasource checks for BOY pvs on startup

berryma4, 2016-10-26:
CS-Studio with only a BOY screen printed the below INFO on startup.  I'm guessing this is the Diirt archive plugin, since it happens at startup.  @jbobnar, I don't see why the plugin would look for non archive:// datasource names?

```
2016-10-26 09:43:04.417 INFO [Thread 282] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - No data found for PV TEST_RB:ISEG_N1101:ON_RSTS + using URL http://epicsarchiver.cts/retrieval/data/getData.raw?pv=TEST_RB%3AISEG_N1101%3AON_RSTS&from=2016-10-25T21%3A43%3A04.336-04%3A00&to=2016-10-26T09%3A43%3A04.336-04%3A00
```

jbobnar, 2016-10-26:
The log was printed from Thread 282. There must be a lot of things going on at startup if that many threads have to be made. It is a query for 12 hours of data. Those 12 hours had to come from somewhere. Diirt archive cannot start loading data on its own.

Can you try to reproduce this from the debugger? There it should be very clear what makes this request.

berryma4, 2016-10-26:
sorry, found it.
Found the databrowser BOY widget in the screen.

closed 2016-10-26 by berryma4