ControlSystemStudio/cs-studio #1943: Check Box widget auto size issue

rosarp, 2016-11-02:
Hi All,

The check box widget has a Label property. It is expected that when the property auto size is checked, changing the length of the label shall resize the widget automatically.
It is not the case.

To resolve this I propose following changes.

```
diff --git a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.widgets/src/org/csstudio/opibuilder/widgets/editparts/CheckBoxEditPart.java b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.widgets/src/org/csstudio/opibuilder/widgets/editparts/CheckBoxEditPart.java
index f79118c..795cd17 100644
--- a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.widgets/src/org/csstudio/opibuilder/widgets/editparts/CheckBoxEditPart.java
+++ b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.widgets/src/org/csstudio/opibuilder/widgets/editparts/CheckBoxEditPart.java
@@ -190,7 +190,7 @@
      * @param figure
      */
     private void performAutoSize(IFigure figure) {
-        getWidgetModel().setSize(((CheckBoxFigure)figure).getPreferredSize());
+        getWidgetModel().setSize(((CheckBoxFigure)figure).getAutoSizeDimension());
     }
 
 
diff --git a/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/CheckBoxFigure.java b/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/CheckBoxFigure.java
index a14789f..cf49d2c 100644
--- a/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/CheckBoxFigure.java
+++ b/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/CheckBoxFigure.java
@@ -21,6 +21,7 @@
 import org.csstudio.swt.widgets.util.GraphicsUtil;
 import org.csstudio.ui.util.ColorConstants;
 import org.csstudio.ui.util.CustomMediaFactory;
+import org.csstudio.ui.util.Draw2dSingletonUtil;
 import org.eclipse.draw2d.ActionEvent;
 import org.eclipse.draw2d.ActionListener;
 import org.eclipse.draw2d.Cursors;
@@ -68,11 +69,13 @@
 
     private Boolean support3d;
 
-    private Color selectedColor=ColorConstants.darkGray;;
+    private Color selectedColor=ColorConstants.darkGray;
+
+    private BoxFigure boxFigure;
 
     public CheckBoxFigure(final boolean runMode) {
         this.runMode = runMode;
-        final BoxFigure boxFigure = new BoxFigure();
+        boxFigure = new BoxFigure();
         setContents(boxFigure);
         if(!runMode)
             setEventHandler(null);
@@ -292,8 +295,17 @@
         this.totalBits = totalBits;
     }
 
+    public Dimension getAutoSizeDimension() {
+        Dimension d = boxFigure.getAutoSizeDimension();
+        int textWidth = d.width();
+        d.setWidth(textWidth + BOX_SIZE + GAP);
+        return d;
+    }
+
 
     class BoxFigure extends Figure{
+        Dimension textSize;
+
         @Override
         protected void paintClientArea(Graphics graphics) {
             if(support3d == null)
@@ -323,7 +335,7 @@
                 });
             }
             graphics.popState();
-            Dimension textSize = FigureUtilities.getTextExtents(text, graphics.getFont());
+            textSize = FigureUtilities.getTextExtents(text, graphics.getFont());
 
             if (!isEnabled()) {
                 graphics.translate(1, 1);
@@ -337,6 +349,28 @@
             super.paintClientArea(graphics);
 
         }
+
+        @Override
+        public Dimension getPreferredSize(int wHint, int hHint) {
+            return getTextSize();
+
+        }
+
+
+        protected Dimension getTextSize() {
+            if (textSize == null)
+                textSize = calculateTextSize();
+            return textSize;
+        }
+
+        protected Dimension calculateTextSize() {
+            return Draw2dSingletonUtil.getTextUtilities().getTextExtents(text, getFont());
+        }
+
+        protected Dimension getAutoSizeDimension(){
+            return getPreferredSize().getCopy().expand(
+                    getInsets().getWidth(), getInsets().getHeight());
+        }
     }
```

Could anyone verify this?

Regards

kasemir, 2016-11-02:
Looks good to me, please create a pull request!

closed 2016-11-08 by berryma4

rosarp, 2016-12-08:
Hi Kay, 
We found that Checkbox label size is not updated sometimes. 
It might happen that checkbox's labe's size is not updated before it executes performAutosize on UI thread with 10ms delay. This delay might not be enough.
Thus forcing its update while updating label text.

Also to note, it is similar to LabelFigure where, while calling setText, textSize should be set to null.

https://github.com/css-iter/cs-studio/blob/5a442e23b61484a2bf0080c2350da284752a4737/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/LabelFigure.java#L223

Creating pull request to fix this.

PR #1947 (2016-11-08): Check Box widget auto size issue
Fixed #1943
PR #1994 (2016-12-08): Checkbox label size is not updated sometimes.
This fixes #1943