ControlSystemStudio/cs-studio #1944: BOY "Open File" action results in changing editor

kasemir, 2016-11-02:
@dylan171 and @berryma4 brought up an issue with the "Open File" action in BOY.
It opens the file with some editor. The specific editor depends on what was used _last_ to open that file, resulting in somewhat unpredictable behavior for the next user on a beamline or in the control room.

The code behind the action is calling `SingleSourceHelperImpl#iOpenFileActionRun`
in the `o.c.opibuilder.rcp` plugin,
which in turn calls `org.eclipse.ui.ide.IDE.openEditor(.., IFile file, ..)`.
That uses `file.getPersistentProperty(EDITOR_KEY)` to determine which editor was used _last_ to open that file.

We could do this to always get the 'default' behavior:
```
  {   // Clear the last-used-editor info to always get the default editor
      file.setPersistentProperty(IDE.EDITOR_KEY, null);
      IDE.openEditor(page, file, true);
  }
```
Now the default editor, which can be configured via Preferences, General, Editors, File Associations, will be used, regardless of which editor was last used in the Navigator via "Open With..".

nickbattam, 2016-11-02:
@kasemir That sounds like a sensible change

dylan171, 2016-11-02:
I think our users would be happy with that change.

PR #1945 (2016-11-02): BOY: Always use default editor for "Open File" action
Excellent comment-to-code ratio of 4:1

Fixes #1944
closed 2016-11-02 by berryma4