ControlSystemStudio/cs-studio #1953: Display Builder question

berryma4, 2016-11-10:
Currently, BOY 'File' -> 'Open' and entering the URL, open the BOY page, but relative paths are lost.  So, if you click a to open a details relative path opi, it can't be found.

Does this "work" in Display Builder?

kasemir, 2016-11-11:
What works:

Define http://path/to/some/file.opi or *.bob for the display builder top displays, then select that from menu File/Top displays (or the toolbar button): Opens in runtime, and relative links are resolved relative to http://path/to/some/

When I use the menu File/Open.., I get a file browser, so I can only select file system files.
When I go to /some/long/path/in/my/workspace/CSS/examples/x.bob, that's opened in the display builder _editor_, and it notices that it's actually in the workspace, so it changes the reference to /CSS/examples/x.bob. So relative paths are OK once you execute it, but I guess you'd like a way to enter any http://.. URL into File/Open, _and_ then get that into the display runtime, not editor?

berryma4, 2016-11-11:
I think it's fine as it is.  The issue is that BOY opens a URL in the File Menu, but doesn't set the relative path.
If Display Builder won't allow a URL in the File Menu, then users won't run into an issue.

Thank you!

closed 2016-11-11 by berryma4

kasemir, 2016-11-11:
How does BOY allow a URL in the file menu?
Are you talking about the menu File/Open?
Maybe this depends on the OS. Right now I'm on Mac OS X, and File/Open presents the OS file system dialog. Then Eclipse takes over and opens the "Editor" for that file.

berryma4, 2016-11-11:
I'm on windows, if you paste a URL in, it takes it.

kasemir, 2016-11-11:
Back home I can try Linux, where it may also allow pasting just anything.

Once the _runtime_ has an http://.. for a file path, the relative links should work.
But I'm not sure how to enforce getting the runtime from File/Open, because it's Eclipse that decides which editor to open, and it tends to open the display _editor_ for me.