ControlSystemStudio/cs-studio #1954: Unable to interact with Text Input in BOY

claudio-rosati, 2016-11-10:
Unzip and import the examples in [opi.zip](https://github.com/ControlSystemStudio/cs-studio/files/583753/opi.zip).

Open with the OPI Display the npmUser.opi file. _Trigger Delay_ was modified to use a local PV.
Clicking on the Text Input doesn't work. Right-click and select _Refresh OPI_: now you can click and edit the value. Clicking again doesn't work. Right-click the Text Input to display the menu without selecting nothing. Now you can click again in the Text Input and edit it.

So after a successful edit, you have to refresh the OPI or just right-click on the Text Input.

Is that a bug or there is some strange and wrong usage of the Text Input settings?

Claudio

kasemir, 2016-11-11:
There is no PV associated with that widget:

[image]

You're right that after some gymnastics like refresh or open-context-menu I can in fact enter a value, but then the widget seems to notice that there really isn't any PV so it goes back into read-only mode.

claudio-rosati, 2016-11-14:
This is very strange: if I set a PV name (e.g. loc://p1), as soon as I save the OPI, it disappear from the widget.

kasemir, 2016-11-14:
Cannot reproduce.
When you set the PV Name, does the *.opi not appear 'dirty' with an '*' in the tab?
Does this only happen when you try to set the PV name of a Text Input which is inside a Tab which in turn is inside a Group etc?

claudio-rosati, 2016-11-15:
I've solved the problem of having the `loc://pv1` name saved: I was editing the main OPI, but this widget is inside a deeper linked container (`npmAssembly.opi`). I really don't like this ambiguity in BOY: you can select a widget into a linked container, changing its properties but then save doesn't apply to it. 
The Display Builder approach is much more consistent. The only thing missing is a command to reload the OPI if changed outside (or is it automatically monitoring for changes in the file?): in BOY a similar command exists in the contextual menu (have we to create a new issue for it?).

So now, I've have my saved local PV, and when I run the main `npmUser.opi` I always have the problem originally described in this issue. The same problem exists also if I run the `npmAssembly.opi` itself.

kasemir, 2016-11-15:
> command to reload the OPI
Display builder is same as BOY: Runtime has a "Re-Load Display" entry in the context menu.

kasemir, 2016-11-15:
As for the original issue: Can you send the simplest possible example that reproduces the error?

claudio-rosati, 2016-11-15:
BOY has a "Re-Load Display" entry in the context menu also in the editor, not only for the runtime.

kasemir, 2016-11-22:
Actually, I don't see a 're-load' in the BOY editor context menu:

[image]

But it is now supported in the display builder editor:

[image]

claudio-rosati, 2016-11-23:
It happens to have it if you right-click a linked container, to update the view of the OPI file changed externally.

[image]

kasemir, 2016-11-23:
Ah, it's only for the linking container. Didn't know that.
Anyway, the display builder editor now has "Reload" in there all the time.
Plus it has the "Execute", which is actually the option that I use a lot.

Do you still see a text entry widget that has a valid, writable PV yet doesn't allow editing?

claudio-rosati, 2016-11-24:
Yes I still have the problem.
I've tried to reproduce it in a simple OPI with just a tab container and a single text input, but I cannot reproduce it.
It seems happening only in the original complex OPI.