ControlSystemStudio/cs-studio #1955: ScanClient API change causing jython scan examples to fail

rosarp, 2016-11-14:
There is a change in API under [#6bd0022991](https://github.com/ControlSystemStudio/cs-studio/commit/6bd0022991ee01783d5e2a04fa7564439e34be52#diff-f49fef019fd53c001fda393021f1be79) revision for ScanClient.

Which is causing [scan_client.py](https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/scan/scan-plugins/org.csstudio.scan/jython/scan_client.py#L131) to fail, since submitScan method now needs 3rd argument.

Based on that also scan examples fails. One of such examples is [custom_scan.py](https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/scan/scan-plugins/org.csstudio.scan/examples/custom_scan.py#L11)

Probable solution for scan_client.py is as follows:

```
diff --git a/applications/scan/scan-plugins/org.csstudio.scan/jython/scan_client.py b/applications/scan/scan-plugins/org.csstudio.scan/jython/scan_client.py
index 80547a4..54003dc 100644
--- a/applications/scan/scan-plugins/org.csstudio.scan/jython/scan_client.py
+++ b/applications/scan/scan-plugins/org.csstudio.scan/jython/scan_client.py
@@ -112,7 +112,7 @@
         return self.client.simulateScan(xml)
 
 
-    def submit(self, name, commands):
+    def submit(self, name, commands, queue):
         """
         Submit a CommandSequence to the server for execution
         
@@ -128,7 +128,7 @@
             xml = commands.getXML()
         else:
             raise Exception('Expecting CommandSequence or XML-text')
-        self.id = self.client.submitScan(name, xml)
+        self.id = self.client.submitScan(name, xml, queue)
         return self.id
 
     def getScanInfo(self, id=-1):
@@ -289,6 +289,9 @@
             args.pop(0)
         else:
             name = "Scan"
+        # This is bool value for queue
+        queue = args[0]
+        args.pop(0)
             
         # End result, overall scan
         cmds = self._decodeScan([], args)
@@ -296,7 +299,7 @@
             raise Exception('Empty scan')
 
         seq = CommandSequence(cmds)
-        id = self.submit(name, seq)
+        id = self.submit(name, seq, queue)
         if __name__ == '__main__':
             seq.dump()
             self.waitUntilDone()
```

kasemir, 2016-11-14:
The org.csstudio.scan/jython/scan_client.py is deprecated.
It's limited to Jython because it depends on calling the Java code for interacting with the scan server.

It's been replaced by the https://github.com/PythonScanClient/PyScanClient library.
The PyScanClient library supports both Jython and C-Python, and has better table scan support.
Its ScanClient.submit call supports the optional queue argument:
````
 def submit(self, cmds, name='UnNamed', queue=True):
 """Submit scan to scan server for execution

        :param cmds: List of commands,
                     :class:`~scan.commands.commandsequence.CommandSequence`
                     or text with raw XML format.
        :param name: Name of scan
        :param queue: Submit to scan server queue, or execute as soon as possible?
        
        :return: ID of submitted scan
        
        Examples::
        
        >>> cmds = [ Comment('Hello'), Set('x', 10) ]
        >>> id = client.submit( cmds, "My First Scan")
        
        >>> cmds = CommandSequent(Comment('Hello'))
        >>> cmds.append(Set('x', 10))
        >>> id = client.submit( cmds, "My Second Scan")
        """
````

So instead of fixing the deprecated code, I'd rather remove it.

kasemir, 2016-11-15:
The org.csstudio.scan/jython/scan_client.py has been adjusted to include the 'queue', but at the same time the examples have been updated to mostly use the PyScanClient examples, removing the older ones.

closed 2016-11-15 by kasemir