ControlSystemStudio/cs-studio #1958: Release 4.4.x (master)

berryma4, 2016-11-15:
Proposing to freeze the master branch to 4.4.x.

Site Representatives, are there any other outstanding issues that need to be addressed before release of 4.3.x?
@berryma4, @kasemir, @shroffk , @utzeln, @mattclarke, @charlesHenryPATARD, @ksdj, @xinyuwu, @jpenning, @willrogers, @claudio-rosati, @gciotto, @wmoore28

jpenning, 2016-11-16:
Hi Eric,

thanks for asking,
currently everything is fine with us 😃

Gruß Jörg Penning. 
(von unterwegs)

charlesHenryPATARD, 2016-11-16:
Hi Eric,
Ok for us too.

Préservons notre environnement, n’imprimez ce mail que si nécessaire.
Preserve our environment, print this email only if necessary.

utzeln, 2016-11-16:
Also OK for ITER. Thanks.

claudio-rosati, 2016-11-16:
OK for ESS too.

Claudio
_____________________________

Claudio Rosati
Software Engineer

European Spallation Source ERIC
P.O Box 176, SE-221 00 Lund, Sweden
Visiting address: Tunavägen 24, Lund

Phone: +46 46 888 34 24
Mobile: +46 721 79 24 24
E-mail: claudio.rosati@esss.se<mailto:petra.ribb@esss.se>

www.esss.se<http://www.esss.se/>

[cid:F13AFA79-E9F7-44E8-95C8-95C364416F06]
_____________________________

EMAIL CONFIDENTIALITY NOTICE

This message is private and confidential.
If you have received this message in error,
please notify me and remove it from your system.

wmoore28, 2016-11-16:
Ok for JLab. 

Wesley

gciotto, 2016-11-16:
Hi Eric,

It's OK for us at LNLS.

berryma4, 2017-01-23:
| Test | Result |
| --- | :-: |
| timestamp | 1/23/2017 15:43:41 |
| issue ticket | https://github.com/ControlSystemStudio/cs-studio/issues/1958 |
| Tester | berryman |
| **BOY Examples** |  |
| All BOY Examples | Pass |
| **Issues** |  |
| Issue 1081 | Pass |
| Issue 1103 | Pass *1 |
| Issue 1197 | Pass |
| Issue 1656 | Pass |
| **Performance Tests** |  |
| graph_performance_test.opi | Pass |
| large_led_performace_test.opi | Pass |
| small_led_performace_test.opi | Pass |
| tank_performace_test.opi | Pass |
| test_update_performance_test.opi.opi | Pass |
| **Rules Tests** |  |
| General_rule.opi | Pass |
| Graphs_rule.opi | Pass |
| LED_color_rule.opi | Pass |
| Label_rule.opi | Pass |
| Switch_rule.opi | Pass |
| **WidgetTests** |  |
| ActionButton_test.opi | Pass |
| BubbleGraph_test.opi | Pass |
| ComboBox_test.opi | Pass |
| Container_test.opi | Pass |
| Gauge_Meter_Thermometer_Tank_test.opi | Pass |
| LED_test.opi | Pass |
| TextInputWidget_test.opi | Pass |
| TextUpdate_test.opi | Pass |
| XYGraph_test.opi | Pass |

*1, Parent_connection_loc.opi gives Warning:
```
2017-01-23 15:40:49 WARNING: Infinite loop was detected when parsing the macro: PRI
```

berryma4, 2017-01-23:
@shroffk do you remember this Warning in the #1103 regression test?

shroffk, 2017-01-23:
```
2017-01-23 15:40:49 WARNING: Infinite loop was detected when parsing the macro: PRI
```
In the Issue1103 test pages I was able to confirm that the Warning is raised when the same marco is used multiple times in the same formula function.

berryma4, 2017-01-23:
closing release as done!!

closed 2017-01-23 by berryma4