ControlSystemStudio/cs-studio #1959: compiling cs-ctudio/core in Eclipse, swt and rap trouble

lotrus, 2016-11-16:
We are working on compiling cs-sutdio/core in Eclipse but we have depedencies error (it's compiling on Maven). Eclipse doesn't find org.eclipse.swt and org.eclipse.rap. We are thinking that we are not finding the good way to declare the plugins in target platform.
Does anyone would have a suggestion or example of target plaform?

kasemir, 2016-11-16:
http://cs-studio.sourceforge.net/docbook/ch04.html
section on "Target Platform" and "Import sources into IDE".

lotrus, 2016-11-16:
Thanks for your help, we have imported Maven Project org.studio.product into Eclipse, we open org.csstudio.product/repository/cs-studio.product

[image]
[image]

 but the Content tab shows features troubles

[image]

When we synchronize and run it, it fails with dependencies trouble:
Framework arguments:  -product org.csstudio.product.product
Command-line arguments:  -product org.csstudio.product.product -data C:\de_filippis_baptiste\workspaceneon2/../runtime-cs-studio.product(1) -dev file:C:/de_filippis_baptiste/workspaceneon2/.metadata/.plugins/org.eclipse.pde.core/cs-studio.product (1)/dev.properties -os win32 -ws win32 -arch x86_64 -consoleLog

!ENTRY org.eclipse.equinox.common 4 0 2016-11-16 16:21:56.996
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: org.eclipse.equinox.common [56]
  Another singleton bundle selected: osgi.identity; osgi.identity="org.eclipse.equinox.common"; type="osgi.bundle"; version:Version="3.8.0.v20160509-1230"; singleton:="true"

       at org.eclipse.osgi.container.Module.start(Module.java:444)
       at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1620)
       at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1600)
       at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1571)
       at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1514)
       at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
       at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
       at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)

kasemir, 2016-11-16:
If you use this approach, the target platform contains all the plugins, but not the features.
So as your screenshot shows, it's missing all the o.c.* features.

I use that approach for a target platform and then import all the CSS sources for core and applications into the IDE, so then the IDE has the features.
If you only want to import the exact sources that you're working on into the IDE, you need a target platform that has all the features which you haven't imported into the IDE.
@shroffk may be able to help with such an approach, I've never used it.

lotrus, 2016-11-17:
We add the plugins locations of the local CSS projects generated by Maven in the Target Platform
[image]
[image]

We have imported the CSS projects in Eclipse (which are compiling with Maven), but we have compiling errors due to dependencies and cs-studio.product could not find the org.eclipse.fx.target.rcp.feature.
[image]

kasemir, 2016-11-17:
> We add the plugins locations of the local CSS projects generated by Maven in the Target Platform

Why did you do that?
You'd obviously only get the binaries for the sources that you already have.
For those, I import the actual sources into the IDE, so I can use them, edit, debug.

What you need in a target platform is the stuff that you _don't_ have available as source code.
The docbook http://cs-studio.sourceforge.net/docbook/ch04.html
section on "Target Platform" and "Import sources into IDE" suggests that you use the "Result of Maven Build", i.e. repository/target/repository of your product build tree.
I use org.csstudio.sns.product/repository/target/repository/plugins.
It has binaries for all the CSS plugins (which I also have as sources, so they're not actually used) plus those plugins that Maven fetched out of the ether for org.apache.batik, com.google.guava, org.antlr.runtime, ..

lotrus, 2016-11-17:
First we tried to put only the org.csstudio.product/repository/target/repository/plugins directory in the target platform as explained in the manual.  But we had many errors.

For example : “The import com.sun.jersey cannot be resolved”.

Then, we noticed that some plugins which were not resolved (like com.sun.jersey), were present in the multiple repository/target/repository obtained after the maven builds. That’s why we tried adding them and it decreased the error number.
But they seem not to be necessary for you.

If we remove them from the target platform, we obtain the errors visible on the screenshots : (“com.sun.jersey cannot be resolved” again, among other ones)
[image]
[image]
[image]

Could our org.csstudio.product/repository/target/repository/plugins be incomplete (org.antlr isn’t there for example)?  Can this be possible even if the maven build worked fine?

kasemir, 2016-11-17:
> Could our org.csstudio.product/repository/target/repository/plugins be incomplete (org.antlr isn’t there for example)?

Yes. I do have org.antlr in the sns.product/repository/target/repository/plugins folder. That explains why your target doesn't work for the IDE. @shroffk, @berryma4 , what do you use for the target, since it's obviously not the SNS product?

kasemir, 2016-12-02:
Any progress?
You could intermediately try building the SNS product as describe on https://github.com/ControlSystemStudio/org.csstudio.sns.
It should "work", you can use the result as a target, it gets you org .antlr etc.
If that still fails, we can compare the log outputs of the various build steps with what I get here where it's building OK.

lotrus, 2016-12-02:
Thanks for asking, we are still working on it.  After building the modules with maven, there were still errors in the “Problems” tab in Eclipse.  So we decided to read all MANIFEST files and added required plug-in that were missing in one of the product feature. The Eclipse's build is successful and our target platform contains now all the required plug-ins.
If we try to “Launch an Eclipse application” through the Overview tab of the .product file, the application doesn’t launch.  But if we go in the “run configuration” menu, click on “add required” and solve validation problems, we are able to launch CSS.
When the application is starting, multiple errors are displayed in the Console.  One of our colleague suggested to uncheck the “rap” plug-ins in the “run configuration” since we are not sure to use it.
If we do so, the number of error or warning are reduced, but still few ones about shortcuts and diirt ones that makes Preference not working well:
```
déc. 02, 2016 9:59:12 AM org.csstudio.diirt.util.Activator start
INFOS: Starting diirt.util
!SESSION 2016-12-02 09:59:10.742 -----------------------------------------------
eclipse.buildId=unknown
java.version=1.8.0_111
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=fr_FR
Framework arguments:  -product org.csstudio.sns.product.product --launcher.defaultAction openFile
Command-line arguments:  -product org.csstudio.sns.product.product -data C:\de_filippis_baptiste\workspacesns/../runtime-sns-css.product -dev file:C:/de_filippis_baptiste/workspacesns/.metadata/.plugins/org.eclipse.pde.core/sns-css.product/dev.properties -os win32 -ws win32 -arch x86_64 -consoleLog --launcher.defaultAction openFile -clean
!ENTRY org.eclipse.equinox.ds 1 0 2016-12-02 09:59:12.037
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterDatasource. The reference is: Reference[name = DataSourceProvider, interface = org.diirt.datasource.DataSourceProvider, policy = dynamic, cardinality = 0..n, target = null, bind = registerDatasource, unbind = deregisterDatasource]
!ENTRY org.eclipse.equinox.ds 1 0 2016-12-02 09:59:12.040
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterFormulaFunctionSet. The reference is: Reference[name = FormulaFunctionSet, interface = org.diirt.datasource.formula.FormulaFunctionSet, policy = dynamic, cardinality = 0..n, target = null, bind = registerFormulaFunctionSet, unbind = deregisterFormulaFunctionSet]
!ENTRY org.eclipse.equinox.ds 1 0 2016-12-02 09:59:12.041
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterService. The reference is: Reference[name = ServiceProvider, interface = org.diirt.service.ServiceProvider, policy = static, cardinality = 0..n, target = null, bind = registerService, unbind = null]
             [NTLoginModule] succeeded importing info:
                    user name = bdefili
 ...
                    user group = S-1-16-8192
                    impersonation token = 2540
**SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [bundleresource://471.fwk1597249648:1/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [bundleresource://471.fwk1597249648:2/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [ch.qos.logback.classic.util.ContextSelectorStaticBinder]**

!ENTRY org.eclipse.jface 2 0 2016-12-02 09:59:14.498
!MESSAGE Keybinding conflicts occurred.  They may interfere with normal accelerator operation.
!SUBENTRY 1 org.eclipse.jface 2 0 2016-12-02 09:59:14.498
!MESSAGE A conflict occurred for F11:
Binding(F11,
       ParameterizedCommand(Command(org.csstudio.opibuilder.actions.fullscreen,Full Screen,
             Enter/Exit Full Screen,
             Category(org.eclipse.ui.category.views,Views,Commands for opening views,true),
             org.eclipse.ui.internal.WorkbenchHandlerServiceHandler@2e426dd2,
             ,,true),null),
       org.eclipse.ui.defaultAcceleratorConfiguration,
       org.eclipse.ui.contexts.window,,,system)
Binding(F11,
       ParameterizedCommand(Command(org.eclipse.debug.ui.commands.DebugLast,Debug,
             Launch in debug mode,
             Category(org.eclipse.debug.ui.category.run,Run/Debug,Run/Debug command category,true),
             org.eclipse.ui.internal.WorkbenchHandlerServiceHandler@1a0ffb9b,
             ,,true),null),
       org.eclipse.ui.defaultAcceleratorConfiguration,
       org.eclipse.ui.contexts.window,,,system)
!SUBENTRY 1 org.eclipse.jface 2 0 2016-12-02 09:59:14.498
!MESSAGE A conflict occurred for CTRL+G:
Binding(CTRL+G,
       ParameterizedCommand(Command(org.csstudio.opibuilder.runopi,Run OPI,
             Run the OPI file currently in the editor,
             Category(org.eclipse.ui.category.edit,Edit,null,true),
             org.eclipse.ui.internal.WorkbenchHandlerServiceHandler@67c0c90b,
             ,,true),null),
       org.eclipse.ui.defaultAcceleratorConfiguration,
       org.eclipse.ui.contexts.window,,,system)
Binding(CTRL+G,
ParameterizedCommand(Command(org.eclipse.jdt.ui.edit.text.java.search.declarations.in.workspace,Declaration in Workspace,
             Search for declarations of the selected element in the workspace,
             Category(org.eclipse.search.ui.category.search,Search,Search command category,true),
[... cut ...]
**2016-12-02 09:59:15.322 AVERTISSEMENT [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Warning: The environment variable HOME is not set. The following directory will be used to store the Git user global configuration and to define the default location to store repositories: 'J:\'. If this is
not correct please set the HOME environment variable and restart Eclipse. Otherwise Git for Windows and
EGit might behave differently since they see different configuration options.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.**
```

We also tried to use the repository/target/products repo as target platform. The result is the same.

shroffk, 2016-12-02:
Hello, 
I am not sure I completely understand your problem, is it to do with some diirt preferences not working or that your IDE has various errors due to missing dependencies (associated with our wonderful target set up process).
If you want we can have a quick hangout and you can walk me through what your trouble with the diirt preferences is.
Cheers, 
Kunal

kasemir, 2016-12-02:
> After building the modules with maven, there were still errors in the “Problems” tab in Eclipse.
> One of our colleague suggested to uncheck the “rap” plug-ins in the “run configuration” since we are not sure to use it.

Yes, when you use the IDE to work on the SWT-based products, I remove all the *.rap.* plugins (and the jre6.fragment which is also for RAP) because otherwise you have the same API twice, once for SWT and once for RAP, and that gets very confusing.

The only errors that remain for me are one on each pom.xml. That's a little inconvenient because it looks like every plugin project is broken on the overview, but when you open a plugin in the navigator the only error is on the pom.xml.

>If we try to “Launch an Eclipse application” through the Overview tab of the .product file, the application doesn’t launch.
>But if we go in the “run configuration” menu, click on “add required” and solve validation problems, we are able to launch CSS.

That depends on the *.product file.
For the SNS products, I added all required plugins to the *.product file, either direct or via the included features. They run from the IDE.
The common CSS product may be minimalistic: It only lists higher level plugins. Maven then includes all the dependencies. The IDE doesn't do that because in the IDE you have the freedom to control which plugins you want to use. Pressing "add required" can help, but it is usually over-eager, adding all *.test fragments that you don't really need.

lotrus, 2016-12-06:
After reading your messages, we tried not to use “add required” button in “run configuration” and launch CSS only with the “Run an Eclipse application” link in the overview tab of the .product. 
We added all the plug-ins that the validation procedure suggested and we managed to launch CSS.

It only remained the problem with diirt.  When we clicked on the “Diirt preferences” tab of the Preferences menu, we had an error message “The currently displayed page contains invalid values”, and then, “platform:/base/diirt“ was displayed in the “Diirt configuration directory” field.  We read on the GitHub that since the last patch about diirt configuration, the diirt preference page was loaded at the preWindowOpen startup stage.  So, we added in pluginCustomization a property "diirt.home=diirt”.
Now CSS is working fine from Eclipse.

Thanks so much for your help!

kasemir, 2016-12-06:
The way diirt is configured is unfortunate. Don't get me started.
It uses XML files outside of Eclipse/CSS.
In Eclipse/CSS we only have a setting for the directory that contains those files.
In the product you use, that is set to "platform:/base/diirt" as a default, meaning:
In the binary product, the one created by Maven, the files will be inside the 'diirt' subdirectory of the generated binary product.
When you execute this product in the IDE, there is no generated binary product, so you need to change that preference to the absolute path of a directory that contains the diirt settings.
In my case, I set it to org.csstudio.sns.product/features/org.csstudio.sns.feature.snscore/rootfiles/diirt because that's the diirt director that gets included in my binary products, so I use the same when testing in the IDE.