ControlSystemStudio/cs-studio #1960: How to have Save and Restore working?

claudio-rosati, 2016-11-16:
Hello,

at ESS we are interested in trying Save and Restore with GIT (first, Masar later).

I've included the 3 *.saverestore*.feature features in our product and rebuilt it.

I've configured it to work with a git repository (remote URL and local path), and I've created a new .bms file inside the repository. The Save Set Editor opens.

Now I've added a description and a single entry into the PV List: `pv1,,,false`. The editor is marked as modified (asterisk in front of the file name in the editor's tab), but I cannot manage to save it. Even closing it and pressing Yes when the 'has been modified' dialog ask for it doesn't work.

I've tried to pud the `pv1,,,false`line into the file with an external editor, but when I double-click it from CS-Studio, the Save Set Editor doesn't display it. It is just empty.

Have you some idea?

Thank you,
Claudio

claudio-rosati, 2016-11-16:
Moreover, when I open the Save & Restore Browser, all the buttons in the Save Sets split and in the Snapshots one are disabled. Is this correct?

berryma4, 2016-11-16:
Maybe a hangout?  So we can go over the workflow?
There is also a help in help contents for the save-restore.
The back end is plug-able.  It might be interesting to add one for an archive appliance (with also using the logbook interface for tagging, searching, etc.)  ... or influxdb (which is popular at the moment).
The git one is nice for starting though.  We use it with our locally installed bitbucket server, which gives us auth.

claudio-rosati, 2016-11-16:
OK.

Tomorrow? (I’m on the bus back home now)

Thank you,
Claudio

_____________________________

Claudio Rosati
Software Engineer

European Spallation Source ERIC
P.O Box 176, SE-221 00 Lund, Sweden
Visiting address: Tunavägen 24, Lund

Phone: +46 46 888 34 24
Mobile: +46 721 79 24 24
E-mail: claudio.rosati@esss.se<mailto:petra.ribb@esss.se>

www.esss.se<http://www.esss.se/>

[cid:F13AFA79-E9F7-44E8-95C8-95C364416F06]
_____________________________

EMAIL CONFIDENTIALITY NOTICE

This message is private and confidential.
If you have received this message in error,
please notify me and remove it from your system.

berryma4, 2016-11-16:
11/17 Thursday 9:00am EST

https://plus.google.com/u/0/b/101349549663920375487/events/cb5np5t8q9to07pg2s45tjcelk8?authkey=COKAz6vUmtexngE

Anyone else interested is free to join.

berryma4, 2016-11-21:
How is this going now?

claudio-rosati, 2016-11-21:
Hi Eric,
I'm working on the DIIRT Preferences Page and I've a talk about the new Display Builder on Wednesday... and no presentation ready yet. I'll work on it on Thursday. I'll let you know.

willrogers, 2016-11-21:
I found out that I was interested in this a few days too late, so sorry for not attending the hangout.

I have exactly the same experience as Claudio - I can't get started.  The help doesn't seem to enlighten me.

berryma4, 2016-11-21:
@willrogers hangout?  Maybe tomorrow 9am EST?

willrogers, 2016-11-21:
Yes, that works for me thanks.  I don't mind updating the help documentation once I've got it figured out.

berryma4, 2016-11-21:
11/22 Tuesday 9:00am EST

https://plus.google.com/u/0/b/101349549663920375487/events/c8tcdiaarn3tf12umur3fa9kh3s?authkey=CP2s68_TxpHRNw

Anyone else interested is free to join.

willrogers, 2016-11-22:
@berryma4 there's a huge meeting in the room I booked for this.  I'm trying to find another room.

berryma4, 2016-11-22:
I got caught up in a meeting ... I will join in 5 minutes

berryma4, 2016-11-23:
#1965 should solve this issue

claudio-rosati, 2016-11-23:
Yes, it works for me now. Thank you.

claudio-rosati, 2016-11-23:
Sorry, I still have the following...

!ENTRY org.eclipse.ui.workbench 4 0 2016-11-23 16:24:31.559
!MESSAGE Invalid thread access
!STACK 0
org.eclipse.swt.SWTException: Invalid thread access
	at org.eclipse.swt.SWT.error(SWT.java:4533)
	at org.eclipse.swt.SWT.error(SWT.java:4448)
	at org.eclipse.swt.SWT.error(SWT.java:4419)
	at org.eclipse.swt.widgets.Widget.error(Widget.java:799)
	at org.eclipse.swt.widgets.Widget.checkWidget(Widget.java:589)
	at org.eclipse.swt.widgets.Widget.getData(Widget.java:865)
	at org.eclipse.ui.internal.ide.handlers.PagePropertyTester.test(PagePropertyTester.java:36)
	at org.eclipse.core.internal.expressions.Property.test(Property.java:58)
	at org.eclipse.core.internal.expressions.TestExpression.evaluate(TestExpression.java:100)
	at org.eclipse.core.internal.expressions.CompositeExpression.evaluateAnd(CompositeExpression.java:53)
	at org.eclipse.core.internal.expressions.WithExpression.evaluate(WithExpression.java:75)
	at org.eclipse.core.internal.expressions.CompositeExpression.evaluateOr(CompositeExpression.java:68)
	at org.eclipse.core.internal.expressions.OrExpression.evaluate(OrExpression.java:22)
	at org.eclipse.core.internal.expressions.ReferenceExpression.evaluate(ReferenceExpression.java:71)
	at org.eclipse.ui.internal.services.EvaluationReference.evaluate(EvaluationReference.java:72)
	at org.eclipse.ui.internal.services.EvaluationReference.evaluate(EvaluationReference.java:97)
	at org.eclipse.ui.internal.services.EvaluationService.requestEvaluation(EvaluationService.java:294)
	at org.eclipse.ui.internal.handlers.DirtyStateTracker.update(DirtyStateTracker.java:45)
	at org.eclipse.ui.internal.handlers.DirtyStateTracker.propertyChanged(DirtyStateTracker.java:107)
	at org.eclipse.ui.part.WorkbenchPart.firePropertyChange(WorkbenchPart.java:120)
	at org.csstudio.saverestore.ui.SnapshotViewerEditor.lambda$1(SnapshotViewerEditor.java:230)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:361)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.BooleanPropertyBase.fireValueChangedEvent(BooleanPropertyBase.java:103)
	at javafx.beans.property.BooleanPropertyBase.markInvalid(BooleanPropertyBase.java:110)
	at javafx.beans.property.BooleanPropertyBase.set(BooleanPropertyBase.java:144)
	at org.csstudio.saverestore.ui.SnapshotViewerController.lambda$0(SnapshotViewerController.java:210)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:328)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:303)
	at org.csstudio.saverestore.SaveRestoreService.setBusy(SaveRestoreService.java:395)
	at org.csstudio.saverestore.SaveRestoreService.setCurrentJob(SaveRestoreService.java:457)
	at org.csstudio.saverestore.SaveRestoreService.access$1(SaveRestoreService.java:454)
	at org.csstudio.saverestore.SaveRestoreService$SaveRestoreJob.run(SaveRestoreService.java:165)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

willrogers, 2016-11-25:
This now seems to be working for me.

berryma4, 2016-11-28:
@claudio-rosati can I close this?

claudio-rosati, 2016-11-28:
Yes, it's OK for me.
(sorry I had a bit complicated days, working on 2 presentations, and few time to test it.)

berryma4, 2016-11-28:
Thank you everyone!!

closed 2016-11-28 by berryma4