ControlSystemStudio/cs-studio #1963: Problem with sun.misc

willrogers, 2016-11-21:
We cannot use pva in our product.  This appears to be related to #1855.

Although `com.diffplug.osgi.extension.sun.misc_0.0.0.jar` is in the plugins directory of the installation, it is not loaded at runtime:

```
osgi> ss misc
"Framework is launched."

id	State       Bundle
```

jbobnar, 2016-11-22:
Try adding this plugin fragment to your product:
https://github.com/ControlSystemStudio/cs-studio/tree/master/core/platform/platform-plugins/org.csstudio.jre6.fragment

willrogers, 2016-11-22:
No, this doesn't seem to help either.  Now `org.csstudio.jre6.fragment_1.0.0.201611211441.jar` is in the built product instead of `com.diffplug.osgi.extension.sun.misc_0.0.0.jar`, but it still isn't loaded when I start CS-Studio:

```
osgi> ss jre
"Framework is launched."

id	State       Bundle
```

willrogers, 2016-11-29:
Adding `com.diffplug.osgi.extension.sun.misc` directly to my product did work.

Should I need to add it to my product or should it be pulled in by a feature?

kasemir, 2016-12-02:
I have it in the SNS product as well.
FWIW, I think the jre6.fragment is only used in RAP products, not SWT products.

willrogers, 2016-12-02:
OK

closed 2016-12-02 by willrogers