ControlSystemStudio/cs-studio #1964: portal "C_n" does not exist - PostgreSQL issue with multiple PVs

rosarp, 2016-11-21:
When multiple PVs are added to the databrowser, few of them fails with exception.
```
WARNING [Thread 220] org.csstudio.trends.databrowser2.ui.Controller$1 (archiveFetchFailed) - No
archived data for CTRL-SUP-BEAU:AI2 org.postgresql.util.PSQLException: ERROR: cursor "<unnamed portal 2>" does not exist
    at org.postgresql.core.v3.QueryExecutorImpl.receiveErrorResponse(QueryExecutorImpl.java:2198)
    at org.postgresql.core.v3.QueryExecutorImpl.processResults(QueryExecutorImpl.java:1927)
    at org.postgresql.core.v3.QueryExecutorImpl.execute(QueryExecutorImpl.java:255)
    at org.postgresql.jdbc2.AbstractJdbc2Statement.execute(AbstractJdbc2Statement.java:561)
    at org.postgresql.jdbc2.AbstractJdbc2Statement.executeWithFlags(AbstractJdbc2Statement.java:405)
    at org.postgresql.jdbc2.AbstractJdbc2Connection.execSQLUpdate(AbstractJdbc2Connection.java:382)
    at org.postgresql.jdbc2.AbstractJdbc2ResultSet.internalGetObject(AbstractJdbc2ResultSet.java:216)
    at org.postgresql.jdbc3.AbstractJdbc3ResultSet.internalGetObject(AbstractJdbc3ResultSet.java:36)
    at org.postgresql.jdbc4.AbstractJdbc4ResultSet.internalGetObject(AbstractJdbc4ResultSet.java:300)
    at org.postgresql.jdbc2.AbstractJdbc2ResultSet.getObject(AbstractJdbc2ResultSet.java:2704)
    at org.postgresql.jdbc2.AbstractJdbc2Statement.executeWithFlags(AbstractJdbc2Statement.java:456)
    at org.postgresql.jdbc2.AbstractJdbc2Statement.execute(AbstractJdbc2Statement.java:412)
    at org.csstudio.archive.reader.rdb.StoredProcedureValueIterator.executeProcedure(StoredProcedureValueIterator.java:130)
    at org.csstudio.archive.reader.rdb.StoredProcedureValueIterator.<init>(StoredProcedureValueIterator.java:68)
    at org.csstudio.archive.reader.rdb.RDBArchiveReader.getOptimizedValues(RDBArchiveReader.java:389)
    at org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread.run(ArchiveFetchJob.java:134)
    at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
    at java.util.concurrent.FutureTask.run(FutureTask.java:266)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
    at java.lang.Thread.run(Thread.java:745)
```

When two PVs are added to the graph, following order of execution causes the exception with PostgreSQL.
Choose PVs with large data and set databrowser to fetch more than 3 months of data.
1. Controller start ArchiveFetchJob AFJ-1 for Pv-1
2. AFJ-1 starts new Worker Thread Wt-1
3. Wt-1 sets autocommit to false
4. On main thread, Controller starts AFJ-2 for Pv-2 
5. AFJ-2 starts new thread Wt-2
6. Wt-2 sets autocommit to false --  [All ok till this]
7. Wt-1 finishes and puts autocommit to true
8. Wt-2 while executing stored procedure, fails since it shares the same connection with Wt-1
9. unnamed portal "C_n" does not exist Exception is thrown - Wt-2 fails to complete

There are two approaches I can think of to solve this issue with "PostgreSQL".
**Approach 1.** To switch ON/OFF by monitoring if any WT is still running. 
Before starting first WorkerThread, set autocommit to false
After all WorkerThreads ends, set autocommit to true.
Need to also consider scenarios with cancel/failure of any queries. And then cleanup the queries.
How it will affect other queries running.. For e.g. WT1 is running, WT2 failed.
We clean up by doing rollback on connection then WT1 will have problems.

**Approach 2.** To have separate connection for each DataSource+PV combination.
Regarding this there were few changes.
Issue - #1286 -  Commit [databrowser: Reduce number of RDB reader JDBC connections]
This commit was made to reduce number of connections. 
Although for PostgreSQL, having separate connections solves the issue.
I suggest _Approach 2_ with following changes.
It will only lock the connection for Postgres, and create new connection for each thread.
It will then release that connection when thread is done using it.
For other Dialect connection will be shared and released when all threads are done using it.
```
diff --git a/applications/archive/archive-plugins/org.csstudio.archive.reader.rdb/src/org/csstudio/archive/reader/rdb/ConnectionCache.java b/applications/archive/archive-plugins/org.csstudio.archive.reader.rdb/src/org/csstudio/archive/reader/rdb/ConnectionCache.java
index f8eaf64..87c4891 100644
--- a/applications/archive/archive-plugins/org.csstudio.archive.reader.rdb/src/org/csstudio/archive/reader/rdb/ConnectionCache.java
+++ b/applications/archive/archive-plugins/org.csstudio.archive.reader.rdb/src/org/csstudio/archive/reader/rdb/ConnectionCache.java
@@ -69,6 +69,8 @@
         private final ID id;
         private final AtomicInteger references = new AtomicInteger(1);
         private final RDBUtil rdb;
+        private int index = 0;
+        private boolean lock = false;

         Entry(final ID id, final RDBUtil rdb)
         {
@@ -92,7 +94,8 @@
     }

     /** Cache */
-    // Expecting only very few entries, so list & linear search good enough
+    // For PostgreSql, need to create separate connections for each thread
+    // For other Dialect, expecting very few entries
     private final static List<Entry> cache = new ArrayList<>();

     /** Find entry
@@ -102,12 +105,30 @@
     private static Entry find(final ID id)
     {
         for (Entry entry : cache)
-            if (entry.id.equals(id))
+        {
+            if (entry.id.equals(id) && entry.lock == false)
             {
                 entry.references.incrementAndGet();
                 return entry;
             }
+        }
         return null;
+    }
+
+    /**
+     *
+     */
+    private static void removeEntry(Entry entry)
+    {
+        int i = 0;
+        for(Entry cachedEntry: cache)
+        {
+            if (entry.index == cachedEntry.index)
+            {
+                cache.remove(i);
+            }
+            i++;
+        }
     }

     /** Get a cached RDB connection
@@ -130,6 +151,13 @@
                 entry = new Entry(id, RDBUtil.connect(url, user, password, false));
                 // Read-only allows MySQL to use load balancing
                 entry.getConnection().setReadOnly(true);
+                // Lock the connection for PostgreSQL, to avoid issue with autocommit with threading
+                // For other Dialects will share connections
+                if(entry.getDialect() == RDBUtil.Dialect.PostgreSQL)
+                {
+                    entry.lock = true;
+                }
+                entry.index = (cache.size() == 0) ? 0 : cache.get(cache.size()-1).index + 1;
                 cache.add(entry);
             }
             return entry;
@@ -145,7 +173,7 @@
         {
             if (entry.references.decrementAndGet() > 0)
                 return; // Still in use
-            cache.remove(entry);
+            removeEntry(entry);
         }
         entry.rdb.close();
         logger.log(Level.FINE, "Closed {0}", entry.id.url);
```

Please suggest which approach is more suitable?

Other related Issue - #1929 - handles cancel of the executing query. Cleans up the connection if aborted/canceled.

kasemir, 2016-11-21:
I agree with your approach of updating the ConnectionCache. For Postgres, it should simply not cache.
Technically, I think this can be done in a simpler way by only modifying the ConnectionCache#release() call. If the dialect is PostgreSQL, ignore the reference count, always remove the entry from the cache and close the connection. So then the next call to get() will always create a new connection. This implementation would not need new index and lock variables and overall be less code.

rosarp, 2016-11-21:
Only release method is not sufficient. 
Since, two threads get connection simultaneously, get(url, user, pass)  method will return same connection for same ID.
Thus, lock was required, to get separate connection for each request.
And index is required to remove correct entry from the List.

kasemir, 2016-11-21:
Good point.
Still, can it be simpler without index and lock?
In get(), if the dialect is Postgres, just don't add the entry to the cache, so it can't be re-used?
In release, cache.remove() will then try to remove an entry that isn't actually in the cache, but List#remove() works OK if the entry doesn't actually exist. It simply returns false, which we never checked anyway.

rosarp, 2016-11-21:
Yes. That should work. By just avoiding cache for PostgreSQL overall it will always create new entry.
So, no overhead at all.. I will try this new solution and then post here my observation.
Thanks.

kasemir, 2016-11-22:
Thanks for fixing it!

PR #1966 (2016-11-23): Postgres avoid connection caching 1964
Fixed #1964
closed 2016-11-22 by kasemir