ControlSystemStudio/cs-studio #1969: compile org.csstudio.product and there are errors,why?

ksspwuwei, 2016-11-25:
[INFO] Reactor Build Order:
[INFO] 
[INFO] product
[INFO] products-csstudio-plugins
[INFO] org.csstudio.product
[INFO] org.csstudio.startup.intro
[INFO] products-csstudio-features
[INFO] org.csstudio.product.feature
[INFO] org.csstudio.product.configuration.feature
[INFO] repository
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building product 4.4.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ product ---
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building products-csstudio-plugins 4.4.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ products-csstudio-plugins ---
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building org.csstudio.product 4.4.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ org.csstudio.product ---
[INFO] 
[INFO] --- tycho-packaging-plugin:0.23.1:build-qualifier (default-build-qualifier) @ org.csstudio.product ---
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] product ............................................ SUCCESS [  0.063 s]
[INFO] products-csstudio-plugins .......................... SUCCESS [  0.003 s]
[INFO] org.csstudio.product ............................... FAILURE [  0.318 s]
[INFO] org.csstudio.startup.intro ......................... SKIPPED
[INFO] products-csstudio-features ......................... SKIPPED
[INFO] org.csstudio.product.feature ....................... SKIPPED
[INFO] org.csstudio.product.configuration.feature ......... SKIPPED
[INFO] repository ......................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 02:28 min
[INFO] Finished at: 2016-11-25T15:47:49+08:00
[INFO] Final Memory: 148M/516M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-packaging-plugin:0.23.1:build-qualifier (default-build-qualifier) on project org.csstudio.product: No git repository found searching upwards from /root/cs-studio/org.csstudio.product/plugins/org.csstudio.product -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :org.csstudio.product

ksspwuwei, 2016-11-25:
I use the command：
[root@wuwei org.csstudio.product]# mvn -B -P my-css-settings clean verify

ksspwuwei, 2016-11-25:
[root@wuwei org.csstudio.product]# mvn -B -P my-css-settings clean verify
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.product:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/plugins/org.csstudio.product/pom.xml
[INFO] Fetching p2.index from http://download.controlsystemstudio.org/updates/4.4/ (0B of 130B at 0B/s)
[INFO] Adding repository http://download.controlsystemstudio.org/updates/4.4
[INFO] Fetching p2.index from http://download.controlsystemstudio.org/core/4.4/ (0B of 130B at 0B/s)
[INFO] Fetching p2.index from http://download.controlsystemstudio.org/applications/4.4/ (0B of 130B at 0B/s)
[INFO] Adding repository http://download.controlsystemstudio.org/issue/1818
[INFO] Adding repository file:/root/cs-studio/my_comp_repo
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/releases/mars
[INFO] Fetching p2.index from http://download.eclipse.org/technology/epp/packages/mars/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201506241002/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201510021000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/releases/mars/201602261000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/eclipse/updates/4.5
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5-201506032000/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.1-201509040015/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.5/R-4.5.2-201602121500/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/rt/rap/3.0
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/R-20150602-0950/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR1a-20150908-1130/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/rt/rap/3.0/SR2-20160201-1220/ (0B at 0B/s)
[INFO] Adding repository http://archive.eclipse.org/rt/rap/incubator/nightly/gef/20150122-1538
[INFO] Fetching p2.index from http://download.eclipse.org/tools/orbit/downloads/drops/R20150519210750/repository/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/tools/orbit/downloads/drops/R20150519210750/repository
[INFO] Adding repository http://download.eclipse.org/efxclipse/runtime-released/2.1.0/site
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.product:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/plugins/org.csstudio.product/pom.xml
[INFO] Resolving class path of MavenProject: org.csstudio:org.csstudio.product:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/plugins/org.csstudio.product/pom.xml
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.startup.intro:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/plugins/org.csstudio.startup.intro/pom.xml
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.startup.intro:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/plugins/org.csstudio.startup.intro/pom.xml
[INFO] Resolving class path of MavenProject: org.csstudio:org.csstudio.startup.intro:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/plugins/org.csstudio.startup.intro/pom.xml
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.product.feature:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/features/org.csstudio.product.feature/pom.xml
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.product.feature:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/features/org.csstudio.product.feature/pom.xml
[INFO] Resolving class path of MavenProject: org.csstudio:org.csstudio.product.feature:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/features/org.csstudio.product.feature/pom.xml
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.product.configuration.feature:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/features/org.csstudio.product.configuration.feature/pom.xml
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.product.configuration.feature:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/features/org.csstudio.product.configuration.feature/pom.xml
[INFO] Resolving class path of MavenProject: org.csstudio:org.csstudio.product.configuration.feature:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/features/org.csstudio.product.configuration.feature/pom.xml
[INFO] Computing target platform for MavenProject: org.csstudio:repository:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/repository/pom.xml
[INFO] Resolving dependencies of MavenProject: org.csstudio:repository:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/repository/pom.xml
[INFO] Resolving class path of MavenProject: org.csstudio:repository:4.4.1-SNAPSHOT @ /root/cs-studio/org.csstudio.product/repository/pom.xml
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] product
[INFO] products-csstudio-plugins
[INFO] org.csstudio.product
[... cut ...]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :org.csstudio.product

kasemir, 2016-11-28:
Looks like your file:/root/cs-studio/my_comp_repo points to things that don't exist, like a repo under /root/cs-studio/org.csstudio.product/plugins/org.csstudio.product that you might be about to create, but it's not there, yet.

I don't know if there are step-by-step instructions on what you should type to build org.csstudio.product from sources.

You can try the instructions for the SNS product, see https://github.com/ControlSystemStudio/org.csstudio.sns#cs-studio-sns .
In includes the maven settings.xml, a composite repo, and all the exact shell commands to build.
Try that, then try to understand it, and then try to do the org.csstudio.product.

ksspwuwei, 2016-11-30:
M2_HOME means what?what should I export it?

kasemir, 2016-11-30:
Before trying to build CSS with Java and Maven, try to familiarize yourself with the basics of those tools. For example, start by googling "M2_HOME" and then for example check http://stackoverflow.com/questions/16223325/purpose-of-the-java-home-m2-and-m2-home-variables. When describing details of the CSS build, we cannot duplicate these basics.

closed 2018-06-12 by berryma4