ControlSystemStudio/cs-studio #1970: Databrowser improvements

willrogers, 2016-11-28:
Our operators have just started using databrowser in earnest and have identified a few problems:

* In Time Axis properties you can edit relative times and the plot will update.  If you edit absolute times nothing happens.
* If absolute times are selected and you press the ... button, both times currently are set to 2016-02-20!
* In Value Axis properties, if you change Min or Max autoscale should be turned off

There are also some feature requests:

* Should PV searching be case-insensitive by default?
* A bunch of possible improvements to the 'Inspect Waveforms' view.

I'm happy to do some of this work, but before we start I'd like to know the plan for databrowser 3:

* Will it reuse any of the current code so we can fix things for both?
* Will it be released very soon so there's no point fixing problems in databrowser 2?

utzeln, 2016-11-28:
We notice that when you edit absolute times, the auto scale of the traces are switched off, which gives often the impression that nothing happens. Actually, if you check the auto scale of each trace, then the plot is displayed correctly.
This happens even if in the preferences, auto size setting is checked.

kasemir, 2016-11-28:
> If absolute times are selected and you press the ... button, both times currently are set to 2016-02-20!

There was a problem with parsing the times and thus showing the wrong start/end time in the dialog because of the time format change. Has been fixed with https://github.com/ControlSystemStudio/cs-studio/pull/1933

kasemir, 2016-11-28:
The case sensitivity of searches is handled by the archive.reader, outside of the data browser 2/3 code. I'm OK with updating that to case-insensitive, but you'd have to do that separately for channel archiver, rdb (likely different SQL dialects), archive appliance, DLS fast/short term archiver, ...

kasemir, 2016-11-28:
As for the databrowser 2/3 plan:
Databrowser 3 is currently in the same repository as the display builder.
When added to the product, you have two toolbar, menu, perspective, .. entries: Original Data Browser and Data Browser 3. For the most part I think it's now doing the same thing, and we just added the display builder widget.
Next would be changing the labels and IDs to match the original DB, so it can be a replacement (but it can then no longer be used at the same time). I expect that to happen by end of January.

willrogers, 2016-11-28:
Is the databrowser 3 code:

* entirely new (completely different bugs)
* partially copied (potentially the same bugs in two places)
* shared (bugs in one place we can fix in one go)

?

(or perhaps
* completely perfect and no bugs, I'm not trying to be rude

)?

kasemir, 2016-11-28:
The new code started with a copy of the existing, then replaced the swt.rtplot with jfx.rtplot, i.e. the plot is based on the JavaFX library. This allows embedding into a display builder widget.
The model and controller part is mostly the same, just modified to support the JFX instead of SWT types of the plot itself.
So I'd say it has the same plus new bugs.
Which is why for now we used different IDs for view, perspective, toolbar etc. to allow running both in parallel to compare and fix things.
Once we match the original IDs, you could only run one at a time. From within the IDE one could have two launch configurations to quickly switch between them, but you could no longer run both concurrently.

willrogers, 2016-11-29:
So I can fix them in Databrowser 2, and you will make sure the fixes are ported (or not necessary) in Databrowser 3?

kasemir, 2016-11-29:
Yes, that'll work.

willrogers, 2017-01-24:
Done with lots of improvements.

closed 2017-01-24 by willrogers