ControlSystemStudio/cs-studio #1973: Invalid thread access from Inspect Waveforms view

willrogers, 2016-11-30:
To reproduce:

* Open databrowser plot
* Right click | Inspect Waveforms
* Add a PV to the databrowser plot

```
2016-11-30 09:38:28.727 WARNING [Thread 74] org.diirt.util.concurrent.Executors$2 (execute) - Exception on the timer thread caused by a ValueListener
org.eclipse.swt.SWTException: Invalid thread access
	at org.eclipse.swt.SWT.error(SWT.java:4491)
	at org.eclipse.swt.SWT.error(SWT.java:4406)
	at org.eclipse.swt.SWT.error(SWT.java:4377)
	at org.eclipse.swt.widgets.Widget.error(Widget.java:482)
	at org.eclipse.swt.widgets.Widget.checkWidget(Widget.java:420)
	at org.eclipse.swt.widgets.Combo.getSelectionIndex(Combo.java:1087)
	at org.csstudio.trends.databrowser2.waveformview.WaveformView.update(WaveformView.java:240)
	at org.csstudio.trends.databrowser2.waveformview.WaveformView.access$0(WaveformView.java:221)
	at org.csstudio.trends.databrowser2.waveformview.WaveformView$1.changedItemLook(WaveformView.java:91)
	at org.csstudio.trends.databrowser2.model.Model.fireItemLookChanged(Model.java:937)
	at org.csstudio.trends.databrowser2.model.ModelItem.fireItemLookChanged(ModelItem.java:198)
	at org.csstudio.trends.databrowser2.model.ModelItem.setUnits(ModelItem.java:169)
	at org.csstudio.trends.databrowser2.model.PVItem.updateUnits(PVItem.java:393)
	at org.csstudio.trends.databrowser2.model.PVItem.pvChanged(PVItem.java:380)
	at org.diirt.datasource.PVReaderImpl.firePvValueChanged(PVReaderImpl.java:110)
	at org.diirt.datasource.PVReaderImpl.setValue(PVReaderImpl.java:220)
	at org.diirt.datasource.PVDirector$4.run(PVDirector.java:358)
	at org.diirt.util.concurrent.Executors$2.execute(Executors.java:55)
	at org.diirt.datasource.PVDirector.notifyPv(PVDirector.java:323)
	at org.diirt.datasource.PVDirector.access$500(PVDirector.java:38)
	at org.diirt.datasource.PVDirector$6.desiredRateEvent(PVDirector.java:403)
	at org.diirt.datasource.SourceDesiredRateDecoupler.sendDesiredRateEvent(SourceDesiredRateDecoupler.java:168)
	at org.diirt.datasource.PassiveScanDecoupler$1.run(PassiveScanDecoupler.java:52)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
```

I'm having a look at this.

willrogers, 2016-11-30:
You can still get the following exception:

```
2016-11-30 15:44:40.246 SEVERE [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Application error
org.eclipse.swt.SWTException: Failed to execute runnable (org.eclipse.swt.SWTException: Widget is disposed)
	at org.eclipse.swt.SWT.error(SWT.java:4491)
	at org.eclipse.swt.SWT.error(SWT.java:4406)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:138)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:3794)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3433)
	at org.eclipse.swt.widgets.Display.release(Display.java:3485)
	at org.eclipse.swt.graphics.Device.dispose(Device.java:248)
	at org.csstudio.startup.application.Application.start(Application.java:129)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:380)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:235)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:669)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:608)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1515)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1488)
Caused by: org.eclipse.swt.SWTException: Widget is disposed
	at org.eclipse.swt.SWT.error(SWT.java:4491)
	at org.eclipse.swt.SWT.error(SWT.java:4406)
	at org.eclipse.swt.SWT.error(SWT.java:4377)
	at org.eclipse.swt.widgets.Widget.error(Widget.java:482)
	at org.eclipse.swt.widgets.Widget.checkWidget(Widget.java:419)
	at org.eclipse.swt.widgets.Combo.setItems(Combo.java:2046)
	at org.csstudio.trends.databrowser2.waveformview.WaveformView.lambda$1(WaveformView.java:230)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:35)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:135)
	... 18 more
```

kasemir, 2016-12-02:
Thanks for fixing this!

closed 2016-12-02 by kasemir