ControlSystemStudio/cs-studio #1977: Sexagesimal number format

lcavalli, 2016-11-30:
Hi,

I need to represent numbers using sexagesimal format.
This function is present in MEDEM, but not in CSS.
I ported MEDEM sexagesimal format from MEDEM to CSS.

[image]

The double to sexagesimal string conversion is a direct Java porting of medemLocalCvtDoubleToSexaStr MEDEM function. The sexagesimal string to double exploits the string split capabilities of Java to reimplement the MEDEM algorithm.

I attach the patch generated with git, but if you are interested in this functionality I can open a pull request.
[sexagesimal.zip](https://github.com/ControlSystemStudio/cs-studio/files/621970/sexagesimal.zip)

berryma4, 2016-11-30:
Thank you!  Please submit a pull request for comments.

lcavalli, 2016-11-30:
Created pull request #1978.

closed 2016-11-30 by berryma4

kasemir, 2016-11-30:
Here's detail from the MEDM manual.
This implementation is similar but expects "00:00:00.000", not allowing those alternate formats with spaces or multiple dots. I think that's fine, even better: Expect the input in same format that's displayed, i.e. with ..:..:..

**sexagesimal**
The text is in degrees or hours, minutes, and seconds with colons between the three fields.  E.g. 12:45:10.2.  The degrees or hours field is the integer part of the number, and the fractional part is expressed as an integer number of minutes and seconds plus fractions of a second.  The example, 12:45:10.2, is the number 12.752833 expressed as 12 degrees or hours, 45 minutes, and 10.2 seconds.  The precision determines the number of digits after the first colon.  For precisions of 0, 1, 2, 3, 4, 5, and 6 the number above would be expressed as 13, 12:5, 12:45, 12:45:1, 12:45:10, 12:45:10.2, and 12:45:10.20, respectively.  On input colons may be replaced by spaces or omitted, and a decimal point may be used.  The inputs 10.75, 10:45, 10.45.0, and 10 45 00 are all treated as a quarter to eleven.  The inputs ‑1, :‑1, and ::‑1 are interpreted as ‑1:00:00, ‑:01:00, and ‑0:00:01, respectively.

**sexagesimal-dms**
This is the same as sexagesimal except that the number is assumed to be in radians and is expressed as degrees, minutes, and seconds with 2p radians representing 360 degrees.  On input the number is multiplied by 180/p and then converted to sexagesimal as above.  On output the number is converted as sexagesimal above and then multiplied by p/180 to get radians.

**sexagesimal-hms**
This is the same as sexagesimal except that the number is assumed to be in radians and is expressed as hours, minutes, and seconds with 2p radians representing 24 hours.  On input the number is multiplied by 12/p and then converted to sexagesimal as above.  On output the number is converted as sexagesimal above and then multiplied by p/12 to get radians.

lcavalli, 2016-11-30:
I also think that having input string equal to representation is better, but if there is consensus I can relax the string parser.