ControlSystemStudio/cs-studio #1980: Feature request: Waveform view should be labelled on main databrowser plot

willrogers, 2016-12-01:
We had a feature request from operators:

> When you have the Inspect Waveform view open, can you tell where on the main databrowser plot you're viewing?

I think this is a good idea, but the question is how best to do it?  I have a prototype that draws a label on the time axis as shown in the picture, and this seems to work quite well.  It requires a bunch of changes around having a new member variable `private XTYPE x_axis_label` in `Plot.java`.

[image]

Thoughts?

kasemir, 2016-12-01:
How about updating the X axis title. Instead of "waveform index", it could show the time stamp of that waveform.

kasemir, 2016-12-01:
Sorry, somewhat misunderstood. You're talking about an indication in the 'source' data browser.

How about adding/moving an annotation? It would better identify which trace and then the position on that trace that's inspected in the waveform view?

willrogers, 2016-12-01:
This was before your new comment, but for the record:

---

The timestamp is actually in the box at the bottom of the waveform view (that's not new).  So the information is there already.

The change is the little label (06:53) on the x-axis of the main plot (just in case you didn't notice that).  The label moves around as you drag the slider.  It might equally be a little red dot on the axis or something else - I just found that a label was easy to draw.

The visual feedback is nice because it is an easier way to understand the link between the two plots.  It also shows you when the data is more or less frequently sampled, which we've already found to be a bit surprising.

The question (as I see it) is whether all the change to the model and plot is worth-while.

willrogers, 2016-12-01:
An annotation sounds like it might be a good idea, plus it would be less invasive to the existing model.  Let me try that.

berryma4, 2016-12-01:
Should I  keep #1981 open for you?

kasemir, 2016-12-01:
Right, the annotation support is already "there", so hopefully this is easier to implement:
Waveform view updates the annotation, and plot already automatically reflects that.

On the other hand, the user can currently move an annotation.
Should this waveform-created annotation be locked from user interaction?
Or should moving the annotation actually select a different instance of the waveform??

willrogers, 2016-12-01:
@berryma4 just for now while we discuss things please.

PR #1997 (2016-12-14): Databrowser waveform view: Second attempt at labelling waveforms with annotations
@willrogers , this is based on your code from #1993, to fix #1980, part of #1970.
Currently selected waveform sample is again indicated by annotation in plot, plus
 * That annotation cannot be deleted or edited by the user, it's not saved in the *.plt file
 * User can move the annotation, whereupon the selected waveform is updated
 * Added flags to break loops between model and plot updates

Not implemented: Some visual indication for offscreen annotations, as the suggested arrow on left resp. right side of plot. [... cut]
closed 2016-12-14 by berryma4