ControlSystemStudio/cs-studio #1985: Databrowser: buffer data vs archive data issue.

berryma4, 2016-12-06:
I have  a user that reported this on a linux machine (cs-studio version 4.3).  Databrowser seems to not grab from the archiver for the buffer length ... which eventually gets filled in.
[image]
[image]

berryma4, 2016-12-06:
I should mention, the same screen works on Windows.

kasemir, 2016-12-06:
Of those 11 channels, I assume the issue is with the top-most wide greenish trace that originally has a larger gap, then later a smaller gap?

Does this data have a gap where a sample is Double.NAN because the archive was off, but the actual gap is shorter than what's initially shown on the screen?
Then this could be the same as #1443: A short gap is erroneously displayed as a much larger one. Was fixed on the master branch on Nov. 13, 2015.

berryma4, 2016-12-06:
It seems to be a threading issue.  It works sometimes, and sometimes not.
It's probably not related to OS.

kasemir, 2016-12-07:
I cannot tell if
a) The wrong archive request is sent out
b) Multiple archive requests are sent out as user zooms/pans around, they're not properly cancelled, so eventually we receive data for the wrong range that no longer matches the plot
c) Data is fine, it's just not fully plotted, maybe because it contains a Double.NaN type gap (#1443)
d) Something entirely different

Enabling the FINE log level shows all the ArchiveFetchJobs (PV name, start and end time).
The "Sample View" shows the current set of samples. That should help to determine what's happening.