ControlSystemStudio/cs-studio #1988: Intermittent deadlock saving databrowser plot to file on Linux

willrogers, 2016-12-07:
* Right-click on a databrowser plot
* Save Snapshot to File ...

About one time in ten, CSS will just hang and need killing. GUI redraws stop happening.

The code is in org.csstudio.swt.rtplot.internal.SnapshotAction.java and I think the offending line is in Plot.java at about line 440:

```
    /** @return {@link Image} of current plot. Caller must dispose */
    public Image getImage()
    {
        Image image = plot_image.orElse(null);
        if (image != null)
            synchronized (image)
            {
                return new Image(display, image, SWT.IMAGE_COPY);   // the problem is here...
            }
        return new Image(display, 10, 10);
    }
```

JConsole does NOT report a deadlock, but it does report that thread 'main' is blocked by 'Image@... owned by: Worker-0'. Worker-0 is waiting on org.eclipse.swt.internal.Lock@....

A possible solution is to put the image data fetch on the UI thread, but this may negate the benefit of having the bulk of the process inside a background job.

Here is a related bug that suggests that this is Linux-specific: https://bugs.eclipse.org/bugs/show_bug.cgi?id=265265

Finally, I realise @kasemir is working on a JFX version that will not be affected.  We do need some fix or workaround (whether that be as simple as disabling this feature) for using databrowser 2 on Linux.

kasemir, 2016-12-07:
If we remove the `synchronized`, the background thread might just then create a new image and call `dispose()` on the one we're holding.

What might work is this non-blocking approach that's aware of the image getting disposed:
````
    public Image getImage()
    {
        while (true)
        {
            Image image = plot_image.orElse(null);
            if (image == null)
                break;
            try
            {
                return new Image(display, image, SWT.IMAGE_COPY);
            }
            catch (Throwable ex)
            {
                // If image was disposed, try again
                // Otherwise give up
                if (! image.isDisposed())
                    break;
            }
        }
        return new Image(display, 10, 10);
    }
````

willrogers, 2016-12-09:
I'm not sure I fully understand.  If the image is disposed, are you confident that another not-disposed image will come along to replace it?

kasemir, 2016-12-09:
Yes. The image is replaced like this:
1) New one created
2) 'image' variable updated to the new image
3) If there was an old image, it's disposed

We synchronized to assert that the image which we received doesn't get disposed while we're looking at it.
Alternatively, we can try to use the image, and if we run into a problem because of step 3, we just try again.

willrogers, 2016-12-12:
But what if the image is genuinely disposed, for example if the plot is closed?

What about (subject to choosing the correct exception)

```
    /** @return {@link Image} of current plot. Caller must dispose */
    public Image getImage()
    {
        Image image = plot_image.orElse(null);
        if (image != null)
        {
            try 
            {
                return new Image(display, image, SWT.IMAGE_COPY);
            }
            catch (Exception e)
            {
                throw new IllegalStateException("Unexpected problem creating plot image.  Please try again");
            }
        }
        return new Image(display, 10, 10);
    }

```
?

This is rare enough that I can't get this failure to happen by my own clicking (I could get the deadlock one if I was patient enough), and the deadlock is definitely the worst-case user experience.

willrogers, 2016-12-12:
This snippet added to `SnapshotAction.java` will reliably recreate the deadlock on Linux.

```
                    try
                    {    // Create snapshot, save to file
                        final ImageLoader loader = new ImageLoader();
                        Image image = null;
                        for (int i = 0; i < 10; i++) {
                            System.out.println("Try number " + i);
                            image = plot.getImage();
                        }
                        final Image image2 = image;
                        loader.data = new ImageData[] { image2.getImageData() };
                        image2.dispose();
                        loader.save(path.get(), SWT.IMAGE_PNG);
                        // Done!
                        break;
                    }
                    catch (Exception ex)
                    {    // Error dialog needs to be in display thread
                        shell.getDisplay().syncExec(() ->
                        {
                            MessageDialog.openError(shell, Messages.Snapshot,
                                    NLS.bind("Cannot save snapshot as {0}.\n\nDetail:\n{1}",
                                             path.get(), ex.getMessage()) );
                        });
                    }
```

kasemir, 2016-12-12:
Looks like we agree that trying again is better than a deadlock.

I would still put the re-try into Plot#getImage(), because it can be called from multiple places: Create image for email, for logbook entry, to save to file, .. Why duplicate the re-try in each caller?

You're correct that this would re-try forever if there's never another image because the plot has been disposed. Very unlikely, because if the plot's been disposed, there won't be any context menu from which to invoke email, logbook, save-image actions. Still, this can be checked:
````
    /** @return {@link Image} of current plot. Caller must dispose */
    public Image getImage()
    {
        // Is there still a canvas that will generate an image?
        if (! isDisposed())
            while (true)
            {
                Image image = plot_image.orElse(null);
                if (image == null)
                    break;
                try
                {
                    return new Image(display, image, SWT.IMAGE_COPY);
                }
                catch (Throwable ex)
                {
                    // If image was disposed, try again
                    // Otherwise give up
                    if (! image.isDisposed())
                        break;
                }
            }
        return new Image(display, 10, 10);
    }
````

willrogers, 2016-12-13:
I was actually just going to give up.  However, this exception is thrown rarely if at all, so I think it makes very little difference.

We'll need to port this back to 4.3 once it's agreed.

willrogers, 2017-01-24:
I hope this is done.

closed 2017-01-24 by willrogers