ControlSystemStudio/cs-studio #1990: Inconsistent text size on TextUpdate widget

mfurseman, 2016-12-08:
We have an issue where the text update widget has inconsistently sized text each time the OPI is refreshed. We've tracked this down to code written to solve https://github.com/ControlSystemStudio/cs-studio/issues/107.

You can reproduce this by creating a text update widget with a small height, and attaching a local enum PV. I've attached the [OPI](https://github.com/ControlSystemStudio/cs-studio/files/639554/TextUpdate.opi.txt)  from the screenshots below.

[image]
[image]

What appears to be happening is that when the OPI is originally drawn the PV is disconnected and the disconnected overlay is drawn at the top of the widget. The code designed to fit text into the widget then tried to resize it to fit into the widget which no longer has a finite size, and only stops reducing the size of the text once the PV is connected. Removing the section of code that does that resizing fixes this issue for us, however I assume that's a breaking change for people who were affected by  https://github.com/ControlSystemStudio/cs-studio/issues/107. Any good ideas for a solution to this?

berryma4, 2016-12-08:
I feel the code written to solve #107 was a pretty bad patch to the larger issue of fonts and screen resolution.  I think @shroffk was more correct with his comment:
https://github.com/ControlSystemStudio/cs-studio/issues/107#issuecomment-22948334

So, it would probably be best to remove the #107 code, and replace it with @shroffk comment.

kasemir, 2016-12-08:
Generally agree that font sizes in the display file should be in pixels, just like all the other sizes and locations in the display file. The display builder uses that approach. Somewhat too late for BOY. Auto-resizing widgets to fit the contained text becomes less important/useful when the font size you see at design time won't change on you, because it's kept in pixels.

So I think I'm OK with removing the resize "fix", but I'm not 100% sure what code we're referring to, so please post a pull request so we see the exact change.

mfurseman, 2016-12-08:
This pull request just removes relevant code, it doesn't attempt to implement @shroffk code yet.

kasemir, 2016-12-08:
Changing BOY to use pixels instead of 'points' for font sizes would be a big change, not worth it.

Just removing that font-shrinking code may also cause quite some upset, as Eric commented.

If I understand correctly, the issue is a combination of this:
1) If the font originally configured is too high, is picks a smaller font.
2) It does that shrinking for any text, including the initial "DISCONNECTED"

-> It shrinks the font once for the initial "DISCONNECTED", but never makes the font grow back up for the later "3.14" text.

==> Would it help to add an is-connected check, and only do the shrinking when connected?

berryma4, 2016-12-08:
Though it is a big change.  I believe it would be worth it.
This has been very difficult to live with, especially with 4k monitors.

mfurseman, 2016-12-08:
I've added a commit that assumes font sizes are in pixels instead of points to the pull request.

@kasemir When the widget is disconnected a white box of a fixed size is drawn onto it (when an actual PV is used, a white border seems to be drawn for local PVs). In our case that white box is larger than the widget itself, so the text is scaled to fit into a box with dimensions [189.0, -14.0].
[image]

kasemir, 2016-12-08:
> added a commit that assumes font sizes are in pixels

What does that do to your existing displays?
I think this should go with an increment of the display file version, and it should only be performed for new files, not when loading files with older version number.

mfurseman, 2016-12-09:
As I think you all expected changing the behaviour but retaining backwards compatibility is quite a lot of work. The changes in the current pull request are enough to get the DPI scaled font, however adding backwards compatibility requires that the TextFigure, and it's subclasses, can be modified from the Model's that actually know what versions they are on, and these things are quite far apart.

willrogers, 2016-12-09:
Matt just talked me through this.  It also seems to me that _given that everything else on OPIs is sized by pixels_ the fonts should also be pixel-based as well, as Kunal suggested.  I don't really see any downsides apart from the obvious one of backward compatibility, which is a big deal.

Could we have a BOY preference **Interpret font sizes in pixels**?  This would allow the default to continue as before but anyone having problems to change the behaviour.

berryma4, 2016-12-09:
Yes, checking the version would be best.  But, it seems a preference would be enough.
I could see someone being annoyed with a 4k monitor, and deciding to change the preference and fix the screen (if needed).  We are really correcting a wrong here, and it's the right way to go.

willrogers, 2016-12-12:
Let's try and summarise these options:

* BOY  preference (user changeable) - a user can change the preference but suddenly half (or all) the screens are slightly wrong
* BOY preference (installation only) - a site has to change the preference all in one go: a lot of effort to get things right
* only new screens get pixel-sized fonts - if you re-save a screen suddenly the fonts change size and need correcting
* new screens get an additional property (pixel-or-point) on text-based widgets.  If the property is not there assume points

The last is the only one that really works.  It requires a change to all text-based widgets.  Does it really need a version check though?

If everyone agrees that this is the correct thing to do then I think we can do it at DLS, although not immediately.

berryma4, 2016-12-12:
Just to be sure, where is this property?
(The last bullet)

willrogers, 2016-12-12:
One property per widget type, alongside the Font property in the Display category.

berryma4, 2016-12-12:
Yes, that sounds perfect.

berryma4, 2017-01-04:
Can I close #1991 ?  Or are you going to put the fix on the same branch/pull request?

willrogers, 2017-01-09:
The new pull request adds a new property wherever there is a font property, which allows you to choose pixels or points.  It defaults to points.  If points, the original resizing behaviour still applies.

There are some things to finish off, but I'd prefer to get some feedback first.

willrogers, 2017-01-09:
I just realised that it would be nicer to incorporate this into `OPIFont.java` and allow it to be selected in the font dialog:

[image]

We could then resize the output in the dialog dynamically to show people what's going on.

I don't know if there would be any problems with this approach.

kasemir, 2017-01-09:
I agree to add this to the OPIFont class.

Can the default be using pixels?
So new displays, and new widgets added to existing displays use pixels.
Old displays are read with points, and when adding a new widget to an old display one can select to still use points.

kasemir, 2017-01-11:
Instead of supporting both pixels and points at the same time with an option to choose either one for each time a font is used, there's another option:

Switch to using pixels, with a scaling factor for fonts in existing displays.
In principle, that scaling factor is just the DPI of the current display.
In reality, it's the DPI of the display that was used to create the display, not necessarily the display that's right now executing the *.opi file, so it needs to be a preference setting.

That's what the display builder does when loading legacy files.

willrogers, 2017-01-11:
Well, I guess I wanted to avoid changing things for existing screens.  If we were going for a full makeover, I think we'd do something like what you describe and not retain the points option.  But I think that would lead to a number of problems for existing sites, especially if they are relying on the existing resize behaviour.

If we retain points, only sites that want to have to make any changes.

Clearly, with these changes a screen could have different types of behaviour associated with different widgets, and that doesn't make a lot of sense, but then there are many ways to make screens that don't make sense.

kasemir, 2017-01-11:
Your approach to still support points is obviously most backwards-compatible, and since you already did the work, no problem.

kasemir, 2017-01-13:
The display builder now recognizes fonts in legacy displays that were stored with pixel-based size.

willrogers, 2017-01-17:
There are a couple of bugs here that I've fixed and I will open a pull request.

I also think we need a BOY preference to determine whether points or pixels are selected by default.

closed 2017-01-23 by berryma4