ControlSystemStudio/cs-studio #1992: BOY Alarm Table widget >> context menu to Create Log Entry & Process Variable

rosarp, 2016-12-08:
At ITER we want to have context menus added to BOY Alarm table widgets.
Particularly 'Create Log Entry' & 'Process Variable' context menus. These are added in BEAST Alarm table using site. Since site is directly available from its view.

But for BOY widgets, OPIRuntimeDelegate has site, and its not passed down to widgets.
It would have been easier if IWorkbenchSite object was available to widgets in AlarmTableWidgetFigure.

To achieve that I have made few changes, to pass on site from OPIRuntimeDelegate to 
WidgetEditPartFactory & through AbstractBaseEditPart, Thus all of its descendants will have access to site, to register context menu.

These changes do not have any impact on BEAST Alarm Table.

To add these menus, I propose below changes.

```
diff --git a/applications/alarm/alarm-plugins/org.csstudio.alarm.beast.ui.alarmtable.opiwidget/src/org/csstudio/alarm/beast/ui/alarmtable/opiwidget/AlarmTableWidgetFigure.java b/applications/alarm/alarm-plugins/org.csstudio.alarm.beast.ui.alarmtable.opiwidget/src/org/csstudio/alarm/beast/ui/alarmtable/opiwidget/AlarmTableWidgetFigure.java
index a73bfad..337e97b 100644
--- a/applications/alarm/alarm-plugins/org.csstudio.alarm.beast.ui.alarmtable.opiwidget/src/org/csstudio/alarm/beast/ui/alarmtable/opiwidget/AlarmTableWidgetFigure.java
+++ b/applications/alarm/alarm-plugins/org.csstudio.alarm.beast.ui.alarmtable.opiwidget/src/org/csstudio/alarm/beast/ui/alarmtable/opiwidget/AlarmTableWidgetFigure.java
@@ -65,9 +65,10 @@
             gui.dispose();
         }
         if (base != null) {
-            AlarmTableWidgetModel model = ((AlarmTableWidgetEditPart) editPart).getWidgetModel();
+            AlarmTableWidgetEditPart widgetEditPart = (AlarmTableWidgetEditPart) editPart;
+            AlarmTableWidgetModel model = widgetEditPart.getWidgetModel();
 
-            gui = new GUI(base, null, model.isWritable(), model.isSeparateTables(), model.getColumns(),
+            gui = new GUI(base, widgetEditPart.getSite(), model.isWritable(), model.isSeparateTables(), model.getColumns(),
                     model.getSortingColumn(), model.isSortAscending(),model.isTableHeaderVisible());
             gui.setNumberOfAlarmsLimit(model.getMaxNumberOfAlarms());
             gui.setBlinking(model.isUnacknowledgedBlinking());
diff --git a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/AbstractBaseEditPart.java b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/AbstractBaseEditPart.java
index 18521e4..113b8b4 100644
--- a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/AbstractBaseEditPart.java
+++ b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/AbstractBaseEditPart.java
@@ -81,6 +81,7 @@
 import org.eclipse.osgi.util.NLS;
 import org.eclipse.swt.SWT;
 import org.eclipse.ui.IActionFilter;
+import org.eclipse.ui.IWorkbenchPartSite;
 import org.eclipse.ui.progress.UIJob;
 
 /**
@@ -116,6 +117,8 @@
     protected Map<String, WidgetPropertyChangeListener> propertyListenerMap;
 
     private ExecutionMode executionMode;
+
+    private IWorkbenchPartSite site;
 
     private TooltipLabel tooltipLabel;
 
@@ -776,6 +779,14 @@
          */
     }
 
+    public IWorkbenchPartSite getSite() {
+        return site;
+    }
+
+    public void setSite(IWorkbenchPartSite site) {
+        this.site = site;
+    }
+
     /**
      * Add/modify an external object from javascript.
      *
diff --git a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/WidgetEditPartFactory.java b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/WidgetEditPartFactory.java
index 06f5ded..319d826 100644
--- a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/WidgetEditPartFactory.java
+++ b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/WidgetEditPartFactory.java
@@ -16,6 +16,7 @@
 import org.csstudio.opibuilder.util.WidgetsService;
 import org.eclipse.gef.EditPart;
 import org.eclipse.gef.EditPartFactory;
+import org.eclipse.ui.IWorkbenchPartSite;
 
 /**The central factory to create editpart for all widgets.
  * @author Sven Wende (class of same name in SDS)
@@ -24,6 +25,13 @@
 public class WidgetEditPartFactory implements EditPartFactory {
 
     private ExecutionMode executionMode;
+
+    private IWorkbenchPartSite site;
+
+    public WidgetEditPartFactory(ExecutionMode executionMode, IWorkbenchPartSite site) {
+        this(executionMode);
+        this.site = site;
+    }
 
     public WidgetEditPartFactory(ExecutionMode executionMode) {
         this.executionMode = executionMode;
@@ -34,8 +42,10 @@
         EditPart part = getPartForModel(model);
         if(part != null){
             part.setModel(model);
-            if(part instanceof AbstractBaseEditPart)
+            if(part instanceof AbstractBaseEditPart) {
                 ((AbstractBaseEditPart)part).setExecutionMode(executionMode);
+                ((AbstractBaseEditPart)part).setSite(site);
+            }
             else if(part instanceof WidgetConnectionEditPart)
                 ((WidgetConnectionEditPart)part).setExecutionMode(executionMode);
         }
diff --git a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/runmode/OPIRuntimeDelegate.java b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/runmode/OPIRuntimeDelegate.java
index ba4ea5e..ba9dae8 100644
--- a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/runmode/OPIRuntimeDelegate.java
+++ b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/runmode/OPIRuntimeDelegate.java
@@ -75,6 +75,7 @@
 
     private IEditorInput editorInput;
 
+    private IWorkbenchPartSite site;
 
     /**
      * The workbench part where the OPI is running on.
@@ -113,6 +114,7 @@
 
     public void init(final IWorkbenchPartSite site, final IEditorInput input) throws PartInitException
     {
+        this.site = site;
         setEditorInput(input);
         if (viewer != null)
             SingleSourceHelper.removePaintListener(viewer.getControl(),errorMessagePaintListener);
@@ -205,7 +207,7 @@
     viewer.createControl(parent);
         viewer.setRootEditPart(root);
         viewer.setEditPartFactory(new WidgetEditPartFactory(
-                ExecutionMode.RUN_MODE));
+                ExecutionMode.RUN_MODE, site));
 
         // viewer.addDropTargetListener(new
         // ProcessVariableNameTransferDropPVTargetListener(viewer));
```

kasemir, 2016-12-08:
@willrogers 
How does with work with the 'shell' mode of the runtime where there is no site?

willrogers, 2016-12-08:
The idea is that all shells are associated with the 'Standalone OPI Summary' view.  The plumbing should be set up so that that view's site is available.

rosarp, 2016-12-13:
This fix is ready, should I create the pull request?

closed 2026-03-09 by rosarp