ControlSystemStudio/cs-studio #1996: RAP for CSS 4.4 opibuilder

MichaelRitzert, 2016-12-08:
A build of the RAP product against the CSS 4.4 p2 repository doesn't start with an error
> Caused by: java.lang.ClassNotFoundException: org.eclipse.swt.SWTError cannot be found by org.eclipse.ui.workbench_3.107.1.v20160120-2131

*swt probably shouldn't be there, so I compared the installed plugins and I found a number of potentially important differences between 4.3 and 4.4.

> org.eclipse.swt
> org.eclipse.swt.gtk.linux.x86
> org.eclipse.text
> org.eclipse.ui
> org.eclipse.ui.ide
> org.eclipse.ui.workbench

are in the 4.4 war, but not in 4.3.

While trying to find out what is pulled in first in the dependency chain, I came across this commit as a possible source: https://github.com/ControlSystemStudio/cs-studio/commit/79b95e1cb0bd95495c3dfdd6c7c78d50869db252. This brings org.eclipse.ui.ide to the opibuilder dependency list when before it was only used by the editor.

kasemir, 2016-12-12:
@willrogers The orgcsstudio.opibuilder.rcp plugin is meant for the code that only works with RCP/SWT, and not RAP. So in principle this  open-in-editor addition to the runtime could be moved into the *.rcp plugin, and the RAP build would then ignore it.

willrogers, 2016-12-13:
OK let me look at this.

kasemir, 2016-12-13:
Issue might be that the org.csstudio.opibuilder.rcp plugin is for RCP/SWT, i.e. non-RAP code for the runtime. But this open-in-editor addition is related to the editor.
So putting it into org.csstudio.opibuilder.rcp might mean that you can no longer build a product that has only the runtime and no editor. I'd be OK with that.

willrogers, 2017-01-04:
The dependencies were removed again in 6d5a9cfb743dd9960187121b6bced3c9e3dab775.

They were moved to `org.csstudio.opibulder`.  I guess that is still a problem.

PR #2029 (2017-01-11): 1996 rap build
@MichaelRitzert @kasemir here is a pull request that attempts to resolve #1996.

I cannot compile RAP, so I don't know that I've done the right thing for the RAP side of this - it may be necessary to return null and handle it rather than throwing the exception.
closed 2017-01-11 by berryma4