ControlSystemStudio/cs-studio #2001: Alarm time since event context menu is empty

rosarp, 2016-12-13:
The context menu from an alarm shows first a clock icon for the time since
event: the elapsed time since the alarm happened.

But the time is empty.

Then clicking on the clock icon, the popup will show the time when the alarm
first occurs and the incomplete text "hh:mm:ss".

Perhaps Duration.between(start, end) is used wrongly in below commit.

https://github.com/ControlSystemStudio/cs-studio/commit/4804cb0ccfab06fe1eaff4d48c9ce8cce32c559e#diff-8d58f04efc6394964b546c3b169ae4f3R86

Same might have happened here..

https://github.com/ControlSystemStudio/cs-studio/commit/4804cb0ccfab06fe1eaff4d48c9ce8cce32c559e#diff-376a49bfb020c842093d3dd84482646dR136

rosarp, 2016-12-21:
@shroffk 

Can you please check this?

shroffk, 2016-12-21:
Thanks, 
I fixed the argument order and added a unit test at least for the NotifierUtil

shroffk, 2016-12-29:
addressed #2021 and #2025

closed 2017-01-13 by shroffk