ControlSystemStudio/cs-studio #2002: Linking Container OPI File property shall support macro

rosarp, 2016-12-13:
In order for the navigation pane to load dynamically only the mimic part of the
canvas, it is required that the linking container shall support macro.

This way, the navigation will update the macro ${OPI_FILE} and refresh the
canvas.
If the macro is inherited from the parent, then it does not work. 

For example:

1. Create A.opi. Add action button ACT_BUTTON.
2. Add Behavior->Actions->Open OPI
File Path = B.opi
Macros = {MIMIC_FILE = B_Mimic}
3. Create B.opi & B_Mimic.opi
4. The macro is used in B.opi in the linking container to load the mimic.
File path = ${MIMIC_FILE}.opi
5. Run A.opi & click on ACT_BUTTON
5. It will load B.opi but will fail to load B_Mimic.opi as the macro inherited from the parent A.opi is not used propeprly by the linking container to load the OPI.

To fix this issue, I propose following changes.

Mainly method fillLinkingContainerSub @@ -495,13 +511,19 @@
needs to be modified such that it fills runtimeMacros to resolve dynamically the opi path. 

Rest of the code is mostly plumbing to pass on macrosInput of parent opi from OPIRuntimeDelegate to XMLUtil.

```
diff --git a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/persistence/XMLUtil.java b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/persistence/XMLUtil.java
index d38b06d..2cb608f 100644
--- a/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/persistence/XMLUtil.java
+++ b/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/persistence/XMLUtil.java
@@ -19,6 +19,7 @@
 import java.util.HashMap;
 import java.util.HashSet;
 import java.util.Iterator;
+import java.util.LinkedHashMap;
 import java.util.List;
 import java.util.Map;
 import java.util.Set;
@@ -184,11 +185,24 @@
      */
     public static void fillDisplayModelFromInputStream(
             final InputStream inputStream, final DisplayModel displayModel, Display display) throws Exception{
-        fillDisplayModelFromInputStreamSub(inputStream, displayModel, display, new ArrayList<IPath>());
+    	fillDisplayModelFromInputStreamSub(inputStream, displayModel, display, new ArrayList<IPath>(), null);
+    }
+
+    /**Fill the DisplayModel from an OPI file inputstream
+     * @param inputStream the inputstream will be closed in this method before return.
+     * @param displayModel. The {@link DisplayModel} to be filled.
+     * @param display the display in UI Thread.
+     * @throws Exception
+     */
+    public static void fillDisplayModelFromInputStream(
+            final InputStream inputStream, final DisplayModel displayModel, Display display,
+            final MacrosInput macrosInput_) throws Exception{
+        fillDisplayModelFromInputStreamSub(inputStream, displayModel, display, new ArrayList<IPath>(), macrosInput_);
     }
 
     private static void fillDisplayModelFromInputStreamSub(
-            final InputStream inputStream, final DisplayModel displayModel, Display display, List<IPath> trace) throws Exception{
+            final InputStream inputStream, final DisplayModel displayModel, Display display, List<IPath> trace,
+            final MacrosInput macrosInput_) throws Exception{
 
         if(display == null){
             display = Display.getCurrent();
@@ -232,7 +246,7 @@
 
         Element root = inputStreamToXML(inputStream);
         if(root != null){
[... 139 lines cut ...]
             }
-
         }
         catch (Exception e)
         {
@@ -396,6 +399,7 @@
 
     private void addRunnerInputMacros(final IEditorInput input) {
         MacrosInput macrosInput = ((IRunnerInput) input).getMacrosInput();
+
         if (macrosInput != null) {
             macrosInput = macrosInput.getCopy();
             macrosInput.getMacrosMap().putAll(
@@ -435,8 +439,9 @@
                                             SingleSourceHelper.removePaintListener(
                                                     viewer.getControl(), loadingMessagePaintListener);
                                         }
+                                        MacrosInput macrosInput = ((IRunnerInput) input).getMacrosInput();
                                         XMLUtil.fillDisplayModelFromInputStream(
-                                                stream, displayModel);
+                                                stream, displayModel, null, macrosInput);
                                         displayModel.setOpiRuntime(opiRuntime);
                                         displayModelFilled = true;
                                         addRunnerInputMacros(input);
```

berryma4, 2016-12-13:
It's best to share a branch, or make a pull request.
This code section is a touchy subject.

berryma4, 2016-12-14:
@shroffk, @willrogers, @kasemir 
Comment?

kasemir, 2016-12-14:
I agree that an embedded display with file path set to "$(FILE)" should resolve macros.
Note that it should NOT react to changes in the macro: Cannot update the macro at runtime and then have the embedded display widget re-load its content.
But should be OK as shown in the example when opening a new display:
Button 1 passes FILE=file1.opi to some display, which then ends up using $(FILE) for the path of an embedded widget -> It shows file1.opi.
Button 2 passes FILE=file1.opi to the same display --> Now it shows file2.opi.

For what it's worth, this does work in the display builder where most properties intrinsically support macros.

The pull request looks OK, but then we thought that before when it comes to embedded displays and macros.
Do you understand why the existing 
````
 String resolvedPath = MacroUtil.replaceMacros(path.toString(), s -> macroMap.get(s));
 path = ResourceUtil.getPathFromString(resolvedPath);
````
isn't sufficient, why a second replacement using
````
  if(path.toString().contains("${") && macrosInput_ != null && macrosInput_.getMacrosMap() != null) {
      	path = container.getOPIFilePath();
      	LinkedHashMap<String, String> runtimeMacros = macrosInput_.getMacrosMap();
      	resolvedPath = MacroUtil.replaceMacros(path.toString(), s -> runtimeMacros.get(s));
       	path = ResourceUtil.getPathFromString(resolvedPath);
````
is required?

This seems to invert the macro hierarchy:
First it applies the macros of the linking container, then, if stuff is still unresolved, it applies the parent macros.
Should be the other way around: First macros from parent, then from the linking container itself which may override some of the inherited macros.
I guess it "works" because the parent macros will only be applied if there's something unresolved, i.e. they won't overrode macros of the linking container itself.

rosarp, 2016-12-19:
Some macros may come at run-time from previously visited OPIs, so those are not available to earlier code.
```
String resolvedPath = MacroUtil.replaceMacros(path.toString(), s -> macroMap.get(s));
 path = ResourceUtil.getPathFromString(resolvedPath);
```
Thus, we needed macrosInput_ at runtime from delegate (OPIRuntimeDelegate) to resolve the macros.

So correct behavior should be ->
We will have macros resolved from parent to children and overridden in children if found the same macro key. 

Is that correct?

kasemir, 2016-12-19:
> We will have macros resolved from parent to children and overridden in children

In principle, that's what I'd call correct.
Similar to environment variables passed into child process, where they can be overwritten.
That's how the display builder, EDM, I think most display tools handle macros.

..  except BOY: The macros on the display itself are used as defaults, for example to allow testing of a display on its own. They are replaced when a display is used as a linked display by the parent macros. So in this case the parent overwrites the macros in the child.
IMHO a bad idea, but some displays may depend on this.

rosarp, 2016-12-29:
@kasemir 
I have updated the code in pull request. 
Could you plesae review it?

It will now always add the macros found in parent to child OPI 
and child OPI will override those if found duplicate.

closed 2026-03-09 by rosarp