ControlSystemStudio/cs-studio #2008: Unit (EGU) can be incorrect if changed on controller

claudio-rosati, 2016-12-14:
This is reported by a colleague at ESS:

> The EGU of the PV can be changed dynamically (in the IOC), but so far I have not found a way to make BOY recognize that change and update the displayed unit.

Is that a bug, or there is some (unknown to me) way to fix the problem?

Thank you,
Claudio

claudio-rosati, 2016-12-14:
Something more about the issue:

>  we came up with a workaround: do not show the units in the TextUpdate of the pressure reading, but create another TextUpdate for the EGU field and place it next to the pressure reading. It is a bit of a work to get the width correct but works..

I don't think that is the right way of working, being TextUpdate handles the unit by itself. So I suppose it's a bug.

kasemir, 2016-12-14:
Historically, Channel Access didn't support updates for metadata (limits, units, precision, ...).

Clients fetched the metadata once on connection, then subscribed to changes in value & alarm state.

Recent EPICS releases support a DBE_PROPERTY subscription to get updates on metadata.
But that caused problems with CA gateways, so it was by default disabled.

==>

You need new IOCs,
a new (or no) gateway,
and enable it via org.csstudio.vtype.pv/support_dbe_property resp. the collection of diirt config files about which you by now know more than I ever wanted to.

kasemir, 2016-12-14:
From EPICS R3.15.0.2 release notes: 
````
Post DBE_PROPERTY events automatically

A new record field attribute "prop(YES)" has been added to identify fields holding meta-data. External changes to these fields will cause a CA monitor event to be sent to all record subscribers who have asked for DBE_PROPERTY updates. Meta-data fields have been marked for all Base record types.
````

kasemir, 2017-02-03:
Suggest to close.
This is not a bug in CSS/pvmanager/vtype.pv/BOY/display builder.
All you have to do is use a recent EPICS release and enable DBE_PROPERTY updates.

claudio-rosati, 2017-02-03:
OK, I've forwarded this issure to the colleague originally raising the problem.

closed 2017-02-03 by claudio-rosati