ControlSystemStudio/cs-studio #2013: Spinner widget continues incrementing when context menu is inadvertently opened

mellguth, 2016-12-20:
In the spinner widget the left mouse button can be hold down on the increment/decrement
buttons to keep incrementing the value. If a user accidentally presses the right mouse button
at the same time, a context menu opens and the spinner widget keeps incrementing even
if the left mouse button is released.
This is dangerous when controlling delicate parts of hardware, especially if the safe operating
range of the variable depends on the situation and therefore has not been statically limited in the 
EPICS IOC / device server.

willrogers, 2016-12-21:
Confirmed on the attached OPI (rename from spinner-opi.txt to spinner.opi to use it).

The button remains down when the left mouse is released - somehow the left-mouse-up cancels the context menu instead of the original left-mouse-down.

[spinner-opi.txt](https://github.com/ControlSystemStudio/cs-studio/files/666656/spinner-opi.txt)

willrogers, 2016-12-21:
There's slightly odd behaviour on the action button widget too.  If you do the same thing, the button remains selected as if you are still clicking on it.  The action doesn't happen in this case.