ControlSystemStudio/cs-studio #2014: Memory leak from Multistate Symbol Widget

willrogers, 2016-12-21:
We tracked down a memory leak to the Multistate Symbol Widget.  It appears to happen when the widget is connected to a PV which has a double value.

The attached screen has many symbol widgets.  Each one is attached to the same DLS PV which has a double value.  The image for each is also attached (I've had to rename them to text files).  I've also attached an image of jconsole showing the memory leak.

[image]

[current-svg.txt](https://github.com/ControlSystemStudio/cs-studio/files/666096/current-svg.txt)

[svg-opi.txt](https://github.com/ControlSystemStudio/cs-studio/files/666094/svg-opi.txt)

boryto, 2017-09-11:
Hi, we tested the provided OPI on latest Cs-studio sources and we could not reproduce the issue. 
The PV that was used had:
 Native data type: DBF_DOUBLE
 Request type:     DBR_DOUBLE

[image]

Can you confirm that this issue is fixed?

mfurseman, 2017-09-14:
It looks like we do still have the memory leak as of https://github.com/ControlSystemStudio/cs-studio/commit/ef16b7e8c88bd574685bb1defd71cc7e54583ac5.

As we seem to be getting different behaviour I wonder if we have different implementations of garbage collection. We're running against Oracle Java 1.8.0.141.x86_64, and only have the two memory arguments set for the VM:

-Xmx3072m
-Xms256

[image]

boryto, 2017-09-14:
Hi, interesting. 
Can you please provide again the test case (was it taken from here?). The IOC is probably somehow simulated. Can you also include the epics db file, so we can have the same PV. 

TNX

mfurseman, 2017-09-14:
We had previously used our beam current PV, which is a real machine PV that updates once a second, for the test. I have just retried it with `sim://sin` and the results are the same. It seems that this may be something more detailed to do with garbage collection though, as the JVM does seem to have freed memory eventually.

[image]

kasemir, 2017-09-14:
What is that tool in the sceenshot? Looks different from jvisualvm and jprofiler, which is what I tend to use.
I agree that details of the GC can be unpredictable. Sometimes the VM will initially use a lot of memory, and after a few hours it decides that much less memory is sufficient.
jvisualvm, jprofiler and also the org.csstudio.utility.sysmon offer a button to trigger GC. If that fails to temporarily free up memory I'd suspect an actual memory leak.

boryto, 2017-09-19:
Hello, 
I can confirm the memory leak. 
[image]
There are way to much org.apache.batik.doom.GenericText objects. Now we need to find the root cause of this. 
TNX

mfurseman, 2017-09-19:
@kasemir, that's jconsole, it seems to be bundled with the Oracle JDK, as well as jvisualvm.

boryto, 2017-10-06:
Reason for memory leak was loading the new image for every different state (every different value is a new state fro image). With double value there were many different states, so many images were loaded for single instance.
This is fixed by rounding the states for images to double and limiting the number of image states to 50.

boryto, 2017-11-14:
The fix is made by rounding the double values to long. Non-integer image states make no-sense.  
This was tested with 100 multistate widgets with sim://sine.  There was no increasing memory over the 1/2h period, and the memory was freed by GC after the OPI was closed. 

[image]

willrogers, 2017-11-15:
Thanks for fixing this. We will test again once this commit is in our build.

closed 2023-10-11 by willrogers