ControlSystemStudio/cs-studio #2017: Databrowser autoscales to all buffered data, not visible data

willrogers, 2016-12-21:
If you set autoscale on a databrowser Y axis, it will autoscale on all data contained in the `PVItem`.  However, this often contains historical data that has scrolled off the left-hand side of the plot, meaning that the autoscale no longer appears sensible.

Could it be changed so that it autoscales on the data visible in the plot?  From poking around I think this would be possible in `PlotProcessor.java`, by passing the x-axis range to `determineValueRange()`.

I can do this if you like.

willrogers, 2017-01-03:
The change would look something like the pull request I just submitted.  It is a behaviour change etc so needs approval.

kasemir, 2017-01-03:
Looks good to me.
Yes, it's a change, but one that matches the expectation of users.
Autoscaling based on some samples that aren't visible wasn't a good idea.
Thanks!

willrogers, 2017-01-24:
Done!

closed 2017-01-24 by willrogers