ControlSystemStudio/cs-studio #2018: Multistate symbol widget throws exceptions when loading certain SVGs

willrogers, 2016-12-21:
I have no idea what's going on here.  Some SVGs (created by Inkscape) cause unhandled exceptions to be thrown by the multistate symbol widget.  The difference between files that do and don't cause the exception are the single line:

```
<svg
  ...
  onload="init(evt)"
  ... >
 ...
</svg>
```

This makes some sense with the inability to load the class Event as in the stack trace.

I have attached two (renamed) SVGs - one with the problem and one without.

[current-ok-svg.txt](https://github.com/ControlSystemStudio/cs-studio/files/666599/current-ok-svg.txt)
[current-exception-svg.txt](https://github.com/ControlSystemStudio/cs-studio/files/666600/current-exception-svg.txt)

```
Exception in thread "Thread-9" java.lang.LinkageError: loader constraint violation: loader (instance of <bootloader>) previously initiated loading for a different type with name "org/w3c/dom/events/Event"
	at java.lang.ClassLoader.findBootstrapClass(Native Method)
	at java.lang.ClassLoader.findBootstrapClassOrNull(ClassLoader.java:1015)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:413)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:435)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:352)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:344)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:160)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.defineClass(ModuleClassLoader.java:272)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.defineClass(ClasspathManager.java:632)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findClassImpl(ClasspathManager.java:588)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClassImpl(ClasspathManager.java:540)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:527)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:324)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:327)
	at org.eclipse.osgi.internal.loader.sources.SingleSourcePackage.loadClass(SingleSourcePackage.java:36)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:386)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:352)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:344)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:160)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.defineClass(ModuleClassLoader.java:272)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.defineClass(ClasspathManager.java:632)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findClassImpl(ClasspathManager.java:588)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClassImpl(ClasspathManager.java:540)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:527)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:324)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:327)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:402)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:352)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:344)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:160)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.apache.batik.dom.svg.SVGDOMImplementation$1.createEvent(SVGDOMImplementation.java:228)
	at org.apache.batik.dom.events.DocumentEventSupport.createEvent(Unknown Source)
	at org.apache.batik.dom.AbstractDocument.createEvent(Unknown Source)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:568)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoadEvent(BaseScriptingEnvironment.java:550)
	at org.apache.batik.bridge.UpdateManager.dispatchSVGLoadEvent(UpdateManager.java:239)
	at org.apache.batik.bridge.UpdateManager.dispatchSVGLoadEvent(UpdateManager.java:220)
	at org.csstudio.utility.batik.SVGHandler$2.run(SVGHandler.java:457)
	at java.lang.Thread.run(Thread.java:745)
```

boryto, 2017-09-11:
Hi, I tested this with latest CSS sources, but could not reproduce the issue. The provided current-exception-svg.txt was used multiple times in OPI. The SVG was used in image and in Image Boolean indicator. 

Can you confirm issue fixed?

boryto, 2017-09-11:
In the latest Cs-studio code I was only able to reproduce org.csstudio.utility.batik.SVGHandler (handleException) - null error, but no Linkage Error. This is probably caused by batik, but need to be investigated more. 

Can you confirm this -- no more Class loading linkage errors?

mfurseman, 2017-09-14:
Hi, I'm working on this with Will at DLS. I can confirm we are seeing the error in our most recent build off https://github.com/ControlSystemStudio/cs-studio/commit/ef16b7e8c88bd574685bb1defd71cc7e54583ac5. Note that the stacktrace is only printed to the system console, not the CS-Studio console.

boryto, 2017-09-14:
Hi, thank you for confirmation. Can we just clarify which error. 
- Exception in thread "Thread-9" java.lang.LinkageError: loader constraint violation: loader (instance of <bootloader>) previously initiated loading for a different type with name "org/w3c/dom/events/Event"
or the 
- org.csstudio.utility.batik.SVGHandler (handleException) - null

Thank you

mfurseman, 2017-09-14:
Hi, the exception I was getting is the LinkageError:
```
Exception in thread "Thread-32" java.lang.LinkageError: org/apache/batik/dom/svg/SVGOMEvent
	at org.apache.batik.dom.svg.SVGDOMImplementation$1.createEvent(SVGDOMImplementation.java:228)
	at org.apache.batik.dom.events.DocumentEventSupport.createEvent(Unknown Source)
	at org.apache.batik.dom.AbstractDocument.createEvent(Unknown Source)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:568)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoadEvent(BaseScriptingEnvironment.java:550)
	at org.apache.batik.bridge.UpdateManager.dispatchSVGLoadEvent(UpdateManager.java:239)
	at org.apache.batik.bridge.UpdateManager.dispatchSVGLoadEvent(UpdateManager.java:220)
	at org.csstudio.utility.batik.SVGHandler$2.run(SVGHandler.java:457)
	at java.lang.Thread.run(Thread.java:748)
```

boryto, 2017-09-19:
Hi, can you also please write the operating system and java version you are using?
TNX

mfurseman, 2017-09-19:
Hi, we are using RHEL6, 2.6.32-696.6.3.el6.x86_64, with Oracle Java 1.8.0.141.x86_64.

boryto, 2017-10-06:
Hi, 
I still cannot reproduce the linkage error. The reason for linkage-error is that are classes loaded form different class loaders.  Seems that my JVM configuration behaves differently or the plugins are loaded in different way (OSGI policy). 

Do you use:
-Dorg.osgi.framework.bundle.parent=ext
-Dosgi.framework.extensions=org.eclipse.fx.osgi

in css.ini?

nickbattam, 2017-10-06:
Yes, we have
```
-Dorg.osgi.framework.bundle.parent=ext
-Dosgi.framework.extensions=org.eclipse.fx.osgi
```
in the css.ini file.

boryto, 2017-10-06:
Can you please also attach the OPI with the svg.
Thank you.

boryto, 2017-10-06:
The attached SVG has onload="init(evt)" without the function defined. Can you also describe the way to reproduce the SVG from Inkscape. 
TNX

willrogers, 2017-10-20:
@gilesknap do you know how these SVGs were made?

gilesknap, 2017-10-27:
I believe  @tomcobb created them in inkscape by hand.

thomascobb, 2017-10-27:
I was trying some animated svgs to see how they looked. Here was the tutorial that I originally followed:

http://www.petercollingridge.co.uk/blog/svg-animation-rotating-elements

I then stripped the animation out, but must have left the init(evt) behind.

willrogers, 2017-10-30:
Arg. In that case, it really looks like the error was on our side.

The widget could perhaps handle this better, but if it's from manual editing of SVG files then it's not so surprising that we run into a problem.

If you think this is going to be difficult to diagnose or fix, then close the ticket and we will re-open it if we find a more general problem.

Sorry that this was so vague and thanks for looking into it.

css-iter, 2017-11-02:
Thank you for your time, so we can have more clear view  about the problem. 

We are a bit limited to batik here with the fix, but we will at least try update the batik version and see if this error is handled better.