ControlSystemStudio/cs-studio #2022: Trouble Running css.exe

WilliamGoode, 2016-12-22:
I am attempting to run css on windows 7. After installing and attempting to run, I am shown the message shown in the attached image. I do not know what this message means or how to resolve the issue. I have verified that I am running Java version 1.8. Any help would be appreciated. 
[image]

berryma4, 2016-12-22:
Where did you download CS-Studio?  Which version?

WilliamGoode, 2016-12-22:
I downloaded it using the following link. I believe it's version 4.1.1

https://ics-web.sns.ornl.gov/css/updates/apps/basic-epics-4.1.1-win32.win32.x86_64.zip

berryma4, 2016-12-23:
64 bit java?

WilliamGoode, 2016-12-23:
32 bit.

berryma4, 2016-12-23:
You downloaded a 64 bit CS-Studio.  You will need the Oracle 64 bit java.

WilliamGoode, 2016-12-23:
That was the problem. Thanks!

closed 2016-12-23 by WilliamGoode