ControlSystemStudio/cs-studio #2023: Overriding managed version 0.23.1 for tycho-p2 plugin.

claudio-rosati, 2016-12-23:
Eclipse Neon shows the "Overriding managed version 0.23.1 for tycho-p2 plugin." warning messages on all `pom.xml` files containing **`<version>${tycho.version}</version>`**:
```
<groupId>org.eclipse.tycho</groupId>
<artifactId>tycho-p2-plugin</artifactId>
<version>${tycho.version}</version>
```
It would be great to revise all the pom files marked as problematic and fix them.

nickbattam, 2016-12-23:
If any change is made to the use of `${tycho.version}` the same logic should be applied to the `tycho-source-feature-plugin`
 ```
   <artifactId>tycho-source-feature-plugin</artifactId>
   <version>${tycho-extras.version}</version>
```

nickbattam, 2016-12-23:
The warning can be suppressed if we resolve the explicit version is required in these files:
```<version>${tycho.version}</version>```

willrogers, 2017-02-23:
I think this is a valid warning.  The Tycho version should be inherited from the parent pom, but we are redefining to (even though we are redefining to the same version).

I think I've fixed all five of these warnings in the cs-studio repo.  Can I close this ticket?

claudio-rosati, 2017-02-23:
Still there in:

- org.csstudio.product
- org.csstudio.display.builder

I've removed the one in org.csstudio.product (see [#24](https://github.com/ControlSystemStudio/org.csstudio.product/pull/24)). For the one in Display Builder I would like to have @kasemir agree on that.

kasemir, 2017-02-23:
Make a pull request for the display builder.

claudio-rosati, 2017-02-23:
OK

claudio-rosati, 2017-02-23:
Done: [#216](https://github.com/kasemir/org.csstudio.display.builder/pull/216)

closed 2020-07-23 by shroffk