ControlSystemStudio/cs-studio #2024: Removing warnings

claudio-rosati, 2016-12-23:
Eclipse Neon is showing 5732 warnings (for all the projects in CS-Studio).
Some of them are described in #2023.
It would be good revising the code/configuration and removing all these warnings.

shroffk, 2016-12-29:
+1
Although I think this has to be dealt with 2 different approaches, while some warning and errors can be addressed  by fixing code there are other that might need to be removed by configuring eclipse preferences.

willrogers, 2017-01-11:
@claudio-rosati have you read this page?  The part about Tycho configurators may reduce some of the warnings you're seeing.

Anyway, please show details of some of the warnings and we can see if some of them can be cleared up.

claudio-rosati, 2017-01-12:
Yew @willrogers, I've read the page on wiki, and everything was already done.

The attached [CR-5732-warnings.txt](https://github.com/ControlSystemStudio/cs-studio/files/701376/CR-5732-warnings.txt) file contains all the warnings I have. I hope you can figure out how to reduce them.

willrogers, 2017-01-12:
I'm generally keen to clean up warnings.  These mostly look genuine, so we should decide how we go about cleaning up.  

The current way we deal with these things is using the checkstyle rules: see https://github.com/ControlSystemStudio/cs-studio/blob/master/build/cs-studio-jenkins-checkstyle

We currently have a few which class as errors (resulting in a failed build) and a few which are just warnings.  An example of what we could do is:

* Remove all the unused imports reported
* Upgrade the warning to error for unused imports

Then the build would fail if anyone tried to commit an unused import, so we'd never get any more.  This would remove 439 of your warnings.  Some of the warnings are more difficult.

After a few false starts, this worked very well for the whitespace problems.

Thoughts @berryma4 @kasemir @shroffk ?

willrogers, 2017-01-12:
Although a lot of the errors in your file come from `org.csstudio.mps.sns`, which is not in the main CS-Studio repo so I don't think is particularly relevant.

claudio-rosati, 2017-01-13:
Thank you @willrogers.

When I was at SNS I added the org.csstudio.sns to Eclipse. I've then removed from the Project view but everything was still in the Navigator one.

Removing it and playing with the Problems view filters (allowing to fix all the warnings in the ESS product and in the diirt project I'm working now), brought me down to 3097.

willrogers, 2017-01-13:
Can you post another file with the remaining errors?

claudio-rosati, 2017-01-13:
Here the file:

[CR-3097-warnings.txt](https://github.com/ControlSystemStudio/cs-studio/files/704124/CR-3097-warnings.txt)

willrogers, 2017-01-13:
"Version is duplicate of parent version" is a Maven warning that is uninteresting to us.

You can disable that with Preferences | Maven | Errors/Warnings and select ignore for that particular error.

I have added this to the wiki page.

willrogers, 2017-01-13:
_GroupId is duplicate of parent groupId_ - we can fix this.

Each pom is specifying `<groupId>org.csstudio</groupId>`.  This is not necessary as it is inherited from the parent pom.  We can remove this in most cases.

willrogers, 2017-01-13:
Of the remaining errors, 2518 are access restriction errors as described by @shroffk in #2032.

That leaves a further 464.  I have categorised them (not completely):

* 128: use of deprecated methods.  These have to be fixed case by case
* 118: bundle cannot be resolved.  This probably depends on the target platform so I'm not sure how to resolve them
* 100: type safety and unchecked casts, use of raw types. I think we could fix most of these
* 58: unused values and methods. We could get rid of these
* 11: no required execution environment.  We can fix this
* 7: unnecessary `@SuppressWarnings`.  We should get rid of these
* About 50 miscellaneous others

willrogers, 2017-02-17:
We could get rid of the vast majority of remaining errors by using a technique like that in #2112.

willrogers, 2017-02-22:
Hi Claudio,

The number of errors on the master branch of cs-studio should be greatly reduced now.  Can you confirm?

Cheers,
Will

claudio-rosati, 2017-02-22:
**applications**: 857 warnings
**core**: 89 warnings
**display builder**: 173 warnings (this is for @kasemir and myself)
**org.csstudio.product**: 4 warnings

Attached the 4 files containing the warnings.

[applications.txt](https://github.com/ControlSystemStudio/cs-studio/files/793358/applications.txt)
[core.txt](https://github.com/ControlSystemStudio/cs-studio/files/793360/core.txt)
[db.txt](https://github.com/ControlSystemStudio/cs-studio/files/793359/db.txt)
[cs-studio-product.txt](https://github.com/ControlSystemStudio/cs-studio/files/793361/cs-studio-product.txt)

willrogers, 2017-02-22:
Damn I missed one.