ControlSystemStudio/cs-studio #2032: Managing eclipse errors and warnings

shroffk, 2017-01-09:
Related to #2024 

In various cs-studio modules imported into eclipse, we are confronted with the access restriction errors
 when the diirt modules are imported in the workspace. (To me these errors seem to not really be valid, the classes are public final)

[image]

The project build path has been configured however, the errors persist.
[image]

willrogers, 2017-02-20:
I don't seem to get these errors.

I've got `org.csstudio.diirt.util` and it seems to link correctly to `org.diirt.datasource` without any errors like that.  They are both imported as Maven projects.  No access rules either.

Do you still see this?

shroffk, 2017-02-21:
I still see this error.
What version of m2eclipse are you using?

shroffk, 2017-02-21:
Currently I use two separate workplaces to overcome this issue.

willrogers, 2017-02-21:
No, I have got it, though I get a warning not an error.

I can't figure it out yet.

This e-mail and any attachments may contain confidential, copyright and or privileged material, and are for the use of the intended addressee only. If you are not the intended addressee or an authorised recipient of the addressee please notify us of receipt by returning the e-mail and do not use, copy, retain, distribute or disclose the information in or attached to the e-mail.
Any opinions expressed within this e-mail are those of the individual and not necessarily of Diamond Light Source Ltd. 
Diamond Light Source Ltd. cannot guarantee that this e-mail or any attachments are free from viruses and we cannot accept liability for any damage which you may sustain as a result of software viruses which may be transmitted in or with the message.
Diamond Light Source Limited (company no. 4375679). Registered in England and Wales with its registered office at Diamond House, Harwell Science and Innovation Campus, Didcot, Oxfordshire, OX11 0DE, United Kingdom

willrogers, 2017-02-22:
This is very strange.  `org.diirt.vtype` is fine but `org.diirt.util` causes problems.  The access rules that you posted a picture of look correct.  However, if you change them, I don't know where the information is stored.

You can get rid of the errors by removing the final 'Forbidden: **/*' from the _vtype_ one, not the diirt util one.  That makes no sense to me.

shroffk, 2017-02-22:
> You can get rid of the errors by removing the final 'Forbidden: **/*' from the vtype one, not the diirt util one. That makes no sense to me.

yes, this is the most confusing thing...I really don't understand this at all.