ControlSystemStudio/cs-studio #2038: OPIBuilder screens react to keyboard presses

willrogers, 2017-01-12:
If you launch an opi screen, one of the buttons will probably be selected so that if you press space or enter, the button will be pressed.  However, there appears to be no way to move the selection to a different button on the page (tab will move it to a different part of the CS-Studio window.

It appears that the first widget in the XML file layout (or equivalently the Outline view) takes the focus.

The problem is that some of our screens have had a motor control as the default selection, and this has caused unintentional motor moves twice now.

Since this appears to be more of a side-effect than a feature, should we disable the selection so that we don't get unintended button presses?

kasemir, 2017-01-12:
Good idea. Technically, you can set the focus. Not sure if you can un-set the focus other than requesting focus for something that's not an *.opi component.

berryma4, 2017-01-12:
I've had users do this.

willrogers, 2017-01-13:
It looks to me like someone (probably Xihui) tried to make this functionality work.  However, I can't manage to traverse the different elements that can be selected.

For example, if you get the Spinner widget selected, you can control it using the up and down keys.  But if you can't get it selected, then you can't.  So unless I'm missing something, this is incomplete.

I think the risk of accidentally moving a spinner outweighs the benefits of being able to use the mouse keys to control it if it is selected.

willrogers, 2017-01-24:
Done.

closed 2017-01-24 by willrogers