ControlSystemStudio/cs-studio #2046: Display Builder, javafx Data Browser cut/paste

berryma4, 2017-01-19:
@dylan171 and I were discussing the new Display Builder and Databrowser.
I know these can be run outside of Eclipse, but the right click to application is gone.
Would it be possible to use the OS clipboard to copy a PV from a widget in Display Builder to the clipboard, then paste it into the standalone javafx databrowser?
Maybe this copy PV from a widget could be a json string with more information {pvname: "testpv", time: "00:00:0000T00:00:00.00"}, so databrowser knew how to handle this json string.
Basically, this would be a simple way to keep the right click ability in the javafx standalone.

Thoughts?

claudio-rosati, 2017-01-19:
The following comes from the Display Builder help page tha I suggest you to give  look to.

**D & D From Outside**

Different entities dragged from outside the editor can be dropped in. When the nature of these entities is compatible, the editor will accept the drag an drop operation creating a series of widgets.

...

**Dropping Text**

A text can be composed by a single or multiple lines, an each line can be interpreted as a PV name. For this reason a dialog will open allowing to select a single Label widget (displaying the whole text), multiple Label Widgets (displaying a line each), or multiple widgets (of the same type) using each line as a PV name.

Multiple widgets will be offset by a grid step in booth coordinates from the preceding one.

kasemir, 2017-01-19:
The data browser is not standalone, neither the original nor the updated one. I assume that would be a lot of work because the data browser is not just the 'editor' for the plot but also a 'property', 'search', 'sample inspect' views which are all SWT and linked via RCP mechanisms.

The display builder runtime is standalone, but as you mention without context menu, because the context menu is in SWT, using RCP to get the PV-related contributions.

Not sure if it's worth implementing what RCP offers outside of RCP.

claudio-rosati, 2017-01-19:
I can add `.json` file support as well as (already planned) a `.template` file support.

berryma4, 2017-01-19:
Ah, I thought databrowser was being rewritten in javafx, I didn't know it wouldn't run standalone.
I do like the idea that the standalone could still pass "some" information.  A json string in the clipboard seems a "neat" way to do that.

kasemir, 2017-01-19:
The underlying plot is JavaFX. This allows using it as a widget in the display builder (or, then wrapped in FXCanvas, as a BOY widget). But that only works for the display builder in Eclipse, since the data browser has too many other RCP dependencies. For example fetching data runs in RCP jobs so they show in the progress view, can be cancelled etc.

berryma4, 2017-01-25:
Closing, I just wanted record Dylan's idea.  I like it because it's a possible interface between other applications (not using RCP).

closed 2017-01-25 by berryma4

kasemir, 2017-01-25:
Understood.

We still need a basic "Copy PV name to clipboard" that only sends "testpv" because that's used a lot:
Get to PV for a `caget` command line to prove that the problem is not in CSS ;-) or to use the PV name in a `grep` of files. In those cases, I don't want `{pvname: "testpv", time: "00:00:0000T00:00:00.00"}` where I then need to delete the JSON wrappings.

In principle, drag data can contain multiple formats: text, image, file reference, .. but support differs in the various APIs. If there was a JSON transfer or XML transfer, we could use that to offer and handle PV + time + display file + ... I don't see that offered in the default  SWT Transfer resp. the JavaFX DataFormat support, but it may still be possible to add a custom transfer type that both can identify and handle.