ControlSystemStudio/cs-studio #2051: New detail panel widget

willrogers, 2017-01-24:
We (DLS) have developed a BOY widget that is like a properties panel but where each row can contain other widgets.  We've been using this widget for a while and now we'd like to share area detector screens that use it.

Can we put the widget in the main repo?  I think it should live in its own plugin, and we can support it.

cc @thomascobb

willrogers, 2017-01-24:
It looks like this:

[image]

berryma4, 2017-01-24:
Yes!

Please make a pull request.

closed 2017-04-04 by willrogers

claudio-rosati, 2017-04-04:
Something similar I'll develop for Display Builder too. Would be interesting be able to import your (BOY) into mine (DB).

willrogers, 2017-04-04:
Yes, that would be very helpful in allowing us to move from BOY to Display Builder.

Thanks!