ControlSystemStudio/cs-studio #2052: Macro substitution in 'Open OPI' action target file

nickbattam, 2017-01-25:
The ActionButton's 'Open OPI' in not able to parse macros for the OPI's file name. 

The use-cases we have are automatic generation of screens where a number of buttons are created to open other OPIs, and re-use of screens across a number of related devices. The most obvious route to pass the target screen is via the macros interface, which is already used for the PV names.

Is there a simple script route to set this property at runtime from a macro defied on the button's OPI? Accessing the macrosMap inside a script does not give access to a macro defined on the button's OPI, so calling `ScriptUtil.openOPI(...)` fails.

In the action, the target file property is explicitly cast to IPath before using it - and behaviour is repeated across all the 'open' actions, which is consistent with their use of the (OPI Path aware) file browser selection of this target.

Updating the action to correctly manage macros looks straight forward, but how to integrate this behaviour with the property in the OPIEditor is less clear.

nickbattam, 2017-01-31:
@rosarp Would the changes you made to LinkingContainer for OPI File macro support (#2005) work for the ActionButton's OPI File handling?

kasemir, 2017-01-31:
I agree that the file name of an open-display-action should support macros.
It does in the display builder.

mfurseman, 2017-07-12:
It seems that this functionality does work, but in order to get the macro based string in the OPI model it must pass the FilePathProperty's [validator](https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/properties/FilePathProperty.java#L88). In this case we can use a file string such as `$(file).opi` and then the string is accepted. This is also the case for the databrowser widget and perhaps others.