ControlSystemStudio/cs-studio #2054: Update to Eclipse 4.6 Neon

kasemir, 2017-01-27:
CSS is still based on Eclipse 4.5 Mars.

What looks interesting from https://www.eclipse.org/eclipse/news/4.6/ :

 * SWT support for high DPI monitors
 * Command to toggle window toolbars, fullscreen option to we can remove similar hacks from BOY & display builder
 * Selecting text editor for all unassociated files
 * Support for Java 9 previews. While Java 9 is delayed, working towards Java 9 #1936 will likely require staying up to date with Eclipse versions.
 * More generics and varargs in RCP API, removing need for instanceof checks and casts
 * Faster Equinox startup (hard to believe)

berryma4, 2017-01-27:
Java 9 feature complete

http://mail.openjdk.java.net/pipermail/jdk9-dev/2017-January/005505.html

Also, openjfx is now in Debian Stretch (and Jessie backports).

boryto, 2017-05-24:
Updated cs-studio repository to compile with Eclipse Neon (4.6.3) plugins.
The branch is available at https://github.com/css-iter/cs-studio/tree/issue_2054.
Updates to latest versions also made for Orbit, Swtboot, and Efxclipse.
For proper navigator icons ApplicationWorkbenchAdvisor.java had to be changed. 
Eclipse sites can also be changed in third-party repository pom.xml.

The build was tested with ITER and general product. Issue 1818 still remains and draw2d has to be changed back to 3.9.101.

kasemir, 2017-05-26:
That doesn't really update the general product because https://github.com/ControlSystemStudio/org.csstudio.product/blob/master/pom.xml still refers to 
```
<eclipse-site>http://download.eclipse.org/releases/mars</eclipse-site>
<eclipse-update-site>http://download.eclipse.org/eclipse/updates/4.5</eclipse-update-site>
```

Same with https://github.com/ControlSystemStudio/org.csstudio.iter/blob/master/pom.xml .

Do you have product build examples that use neon, successfully?

boryto, 2017-05-29:
Yes, the product pom.xml has to be updated. We successfully build the general and the ITER product.  The first tests introduced the small navigator icons problem which was fixed in   ApplicationWorkbenchAdvisor.java

Major tests on ITER product are in progress, no big issues so far.

boryto, 2017-05-31:
The latest ITER sources were pushed to the https://github.com/ControlSystemStudio/org.csstudio.iter

Shall we update also the general product with neon and other updates?

kasemir, 2017-05-31:
Yes, please update the general product.

berryma4, 2017-06-02:
@boryto were you going to update the general product?

claudio-rosati, 2017-06-02:
I have a lot of problems having my build scripts working after 4.6 updates (see #2210). 

After the first merges relating to Neon 4.6 my build scripts stopped to work. Thank to @boryto I was upble to manage `cs-studio-thirdparty`, `cs-studio/core` and `cs-studio/applications`, but it still impossible have `org.csstudio.display.builder` built and artifacts being created for  `org.csstudio.product`.

boryto, 2017-06-05:
Pull request created for general product. https://github.com/ControlSystemStudio/org.csstudio.product/pull/27

kasemir, 2017-06-13:
The SNS products all compile, no known issues.
The 4.6.3 IDE, under Linux, keeps showing a "PartRenderingEngine's limbo" window. Looks like that's a bug known for two years, https://bugs.eclipse.org/bugs/show_bug.cgi?id=473278 . Haven't seen that on beamlines in the CSS products based on 4.6, yet.

claudio-rosati, 2017-06-14:
All ESS products compile and all JIRA jobs are working well now.

The only problem is a the following exception appearing twice (not being there before):

```
2017-06-14 16:17:43.009 SEVERE [Thread 24] org.csstudio.logging.PluginLogListener (logging) - Plug-in org.eclipse.egit.ui was unable to load class org.eclipse.egit.ui.internal.decorators.GitLightweightDecorator.
java.lang.ClassNotFoundException: An error occurred while automatically activating bundle org.eclipse.egit.ui (285).
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:116)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:529)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:325)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:345)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:423)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.loadClass(EquinoxBundle.java:564)
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.createExecutableExtension(RegistryStrategyOSGI.java:174)
	at org.eclipse.core.internal.registry.ExtensionRegistry.createExecutableExtension(ExtensionRegistry.java:905)
	at org.eclipse.core.internal.registry.ConfigurationElement.createExecutableExtension(ConfigurationElement.java:243)
	at org.eclipse.core.internal.registry.ConfigurationElementHandle.createExecutableExtension(ConfigurationElementHandle.java:55)
	at org.eclipse.ui.internal.services.WorkbenchServiceRegistry.getSourceProviders(WorkbenchServiceRegistry.java:177)
	at org.eclipse.ui.internal.services.SourceProviderService.readRegistry(SourceProviderService.java:108)
	at org.eclipse.ui.internal.Workbench$47.runWithException(Workbench.java:2447)
	at org.eclipse.ui.internal.StartupThreading$StartupRunnable.run(StartupThreading.java:32)
	at org.eclipse.swt.widgets.Synchronizer.syncExec(Synchronizer.java:233)
	at org.eclipse.ui.internal.UISynchronizer.syncExec(UISynchronizer.java:145)
	at org.eclipse.swt.widgets.Display.syncExec(Display.java:4814)
	at org.eclipse.ui.internal.StartupThreading.runWithoutExceptions(StartupThreading.java:95)
	at org.eclipse.ui.internal.Workbench.initializeDefaultServices(Workbench.java:2442)
	at org.eclipse.ui.internal.Workbench.init(Workbench.java:1678)
	at org.eclipse.ui.internal.Workbench.runUI(Workbench.java:2944)
	at org.eclipse.ui.internal.Workbench.access$9(Workbench.java:2875)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:684)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:103)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
Caused by: org.osgi.framework.BundleException: Exception in org.eclipse.egit.ui.Activator.start() of bundle org.eclipse.egit.ui.
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:795)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:724)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:932)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:309)
	at org.eclipse.osgi.container.Module.doStart(Module.java:581)
	at org.eclipse.osgi.container.Module.start(Module.java:449)
	at org.eclipse.osgi.framework.util.SecureAction.start(SecureAction.java:470)
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:107)
	... 44 more
Caused by: java.lang.LinkageError: loader constraint violation: when resolving method "org.slf4j.impl.StaticLoggerBinder.getLoggerFactory()Lorg/slf4j/ILoggerFactory;" the class loader (instance of org/eclipse/osgi/internal/loader/EquinoxClassLoader) of the current class, org/slf4j/LoggerFactory, and the class loader (instance of org/eclipse/osgi/internal/ [... cut ...]
	at org.eclipse.jgit.lib.Repository.<clinit>(Repository.java:110)
	at org.eclipse.egit.ui.Activator.setupRepoIndexRefresh(Activator.java:423)
	at org.eclipse.egit.ui.Activator.start(Activator.java:307)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:774)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:767)
	... 51 more
```

kasemir, 2017-06-14:
My guess would be that something other than the git support had called slf4j earlier. Does your product contain more than one org.slf4j.* plugin?
For what it's worth, the SNS product includes jgit/egit, the git UI works, and I don't see these slf4j error messages. You can see the build settings at https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build and the latest output at https://ics-web.sns.ornl.gov/css/nightly/

claudio-rosati, 2017-06-16:
I found the problem in my product's feature file. Thank you.

willrogers, 2017-06-29:
I realise this might make people tear their hair out, but...

Since Eclipse Oxygen has just been released, should we also jump to that while we are resolving compatibility problems?

berryma4, 2017-06-29:
Let's discuss this at our next meeting.

willrogers, 2018-03-09:
This is done and I'm looking at Oxygen. Close?

closed 2018-03-09 by berryma4