ControlSystemStudio/cs-studio #2055: Few suggestions about BOY/Display Buider from an external view

pavel-ch, 2017-01-29:
As agreed in https://github.com/kasemir/org.csstudio.display.builder/issues/188, I put down few suggestions according BOY/Display Builder:
1.(MAJOR)  Implement more application (BOY/Display Builder) control over display windows positioning and attributes like window constraints (fixed size, fixed position,  no minimize, always on top, position relative to mouse click etc.). 
2. (MINOR) Implement user rights, roles etc for BOY/Display Builder
3. (MINOR) Make Linking Container with transparent background option

The whole text you can read here:
https://docs.google.com/document/d/1VD6p2J8kvC4NDVUn-wFoWmOF1hiJvRqI7kJcINOGjK8

br Pavel

willrogers, 2017-01-30:
Thanks for the detailed feedback, it shows a good understanding of CS-Studio.

Re 1, you have correctly identified RCP as both an advantage and a limitiation of CSS as it is now.  My view is no different to yours:

* it enables a lot of the integration that you get in CSS
* it presents restrictions that are awkward to get around

It is possible to adapt RCP closer to requirements like yours, slowly, with a fair amount of work.  But it presents difficulties in keeping the interface intuitive.  You have seen this in OPIShell.  One relevant question is whether Kay intends to keep the ability to launch Display Builder independently of CSS itself.

For 2 and 3, these are perfectly sensible (I think) requirements that would usually be implemented by a site that identified the need.

One general question for Pavel: it appears natural for these control screens eventually to be available as HTML, solving a number of platform issues.  How would this fit with your experience in industry?

berryma4, 2017-01-30:
Thank you very much for your time, these are great suggestions.
We have discussed limitations and advantages of using Eclipse RCP informally, maybe it's time to reevaluate our use, or how we use it.  I don't mean to pick on Eclipse RCP, any framework will have limitations.